project-chip/connectedhomeip #17209: RemoveTrustedRootCertificate does not work according to spec

tcarmelveilleux, 2022-04-08:
#### Problem
The `RemoveTrustedRootCertificate` handler in the SDK does not seem to work at all.

By inspecting the code, we find:

```
// TODO: Implement the logic for RemoveTrustedRootCertificate
    EmberAfStatus status = EMBER_ZCL_STATUS_FAILURE;
    emberAfSendImmediateDefaultResponse(status);
```

 #### Proposed Solution
Fix the logic of RemoveTrustedRootCertificate to follow spec. This is symmetrical to #17208

tcarmelveilleux, 2022-05-25:
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/5269 covers a fix in spec to match SDK by removing the command

woody-apple, 2022-05-26:
Spec Review: After the above spec change, this is not a spec blocking issue.

tcarmelveilleux, 2022-05-26:
Spec changes impact AddTrustedRootCertiticate as well

PR #18965 (2022-06-01): Multiple spec compliance fixes to NOC cluster
#### Problem  
- A lot of error-handling in NOC cluster is not spec-compliant with too many different situations to list individually.
- The RemoveTrustedRootCertificate command no longer exisets
- There is missing interlock between the fail-safe and AddNOC/UpdateNOC/AddTrustedRootCertificate/CSRRequest

Issue #18633
Fixes #15311
Fixes #17209

#### Change overview
- Add all missing IM-level checks that would lead to
  INVALID_COMMAND error on incorrectly provided inputs
- Normalize logging of errors some more
- Add missing interlocks on fail-safe
- [... cut]
commit 50d1c8e3 (2022-06-01): Multiple spec compliance fixes to NOC cluster (#18965)

* Multiple spec compliance fixes to NOC cluster

- Add all missing IM-level checks that would lead to
  INVALID_COMMAND error on incorrectly provided inputs
- Normalize logging of errors some more
- Add missing interlocks on fail-safe
- Remove RemoveTrustedRootCertificate command
- Ensure error-handling in AddTrustedRootCertificate
  does not destroy in-progress state on innocuous errors
- Move some IM structural checks before fail-safe checks
  in AddNOC/UpdateNOC

Issue #18633
Fixes #15311
Fixes #17209

* ZAP changes

* Restyled by clan [... cut]
closed 2022-06-01 by tcarmelveilleux