project-chip/connectedhomeip #17221: When the NOC for a fabric changes, we need to invalidate all CASE resumption state for the fabric.

bzbarsky-apple, 2022-04-08:
#### Problem
There are two ways the NOC can change:

1. `UpdateNOC` command processing.
2. A controller being set up with a new NOC that does not match the one it used to have before (see `DeviceController::ProcessControllerNOCChain`.  This is the case when `mFabricInfo = params.systemState->Fabrics()->FindFabric(rootPublicKey, fabricId);` returns a non-null entry but the passed-in NOC does not match the one in the table.  Maybe in this case we can assume that the relevant bits got removed from storage anyway, but it might be better not to...

Relevant spec bit is in "11.18.7.9. UpdateNOC Command" and says:

> All internal data reflecting the prior operational identifier of the Node within the Fabric SHALL be revoked and removed, to an outcome equivalent to the disappearance of the prior Node, except for the ongoing CASE session context, which SHALL temporarily remain valid until the NOCResponse has been successfully delivered or until the next transport-layer error, so that the response can be received by the Administrator invoking the command.

(thanks to @tcarmelveilleux  for the spec pointer).

 #### Proposed Solution
As a bare minimum, need to remove the relevant CASE resumption bits from storage when `UpdateNOC` happens.  Ideally the same during controlller init.

@kghost

kghost, 2022-04-08:
I suggest that the safest path is saving the NoC fingerprint in the resumption state storage and comparing it to check if it is changed when loading it. In this way, we can avoid problem like power cut while removing the entry.

bzbarsky-apple, 2022-05-13:
I was thinking about how to handle this a bit, and fundamentally we have no good way (especially on the controller side) of telling when the NOC has changed across a reboot.  Whatever comes out of storage might just be different, and all we will validate in FabricTable is that it chains up to the root properly..

So something like what @kghost suggested might in fact be the simple way to go: saving the NOC public key in the resumption state and then making sure it matches

bluebin14, 2022-05-14:
Note that the only thing that causes TC-IDM-1.2 and other tests to fail is that in case of resumption the device has no knowledge that the controller nodeid changed, it uses the old nodeid and that causes decryption to fail. Nodeid is used to construct the crypto nonce. Resumption state needs to know the nodeids of both parties, if any of them changes the state becomes invalid.

bzbarsky-apple, 2022-05-14:
You'd also get failures if the operational key changed, even if the node id did not change, right?

bluebin14, 2022-05-14:
There would be no decryption failures in that case.

bluebin14, 2022-05-14:
The problem here was that both sides acknowledged successful session completion yet they could not communicate because of crypto failure.

bluebin14, 2022-05-14:
Note also that e.g. chip-tool changes its own keypair at every launch, so invalidating resumption state on public key change on either side would cause resumption never to be used.

woody-apple, 2022-05-19:
Cert Blocker Review: @mrjerryjohns is also working on this, assigning.

mrjerryjohns, 2022-06-01:
@kghost please confirm you can help take a look at this.

kghost, 2022-06-01:
will look it today

andy31415, 2022-06-08:
Sync with @kghost result: fevelopment of this depends on work from @tcarmelveilleux on NOC. Unassigning until development can start.

Will have to clarify dependencies here.

tcarmelveilleux, 2022-06-08:
> Development of this depends on work from @tcarmelveilleux on NOC. Unassigning until development can start.

The dependency is that my UpdateNOC work on #18633 would use this API if it was there. This work, however is can begin without conflicts

woody-apple, 2022-06-13:
Cert Blocker Review: Needs an assignee.

closed 2022-06-24 by tehampson