project-chip/connectedhomeip #17235: iOS project is not compiling

Lee6804, 2022-04-11:
Package openssl was not found in the pkg-config search path.
Perhaps you should add the directory containing `openssl.pc'
to the PKG_CONFIG_PATH environment variable
No package 'openssl' found
Could not run pkg-config.

stlevkov, 2022-04-11:
Hi, 
try to find openssl.pc on your host machine. If it's missing, probably the package needs to be installed. Then if the compiler still cannot resolve it, add the dir containing the .pc file to the PKG_CONFIG_PATH.

closed 2022-04-12 by Lee6804

andrewstang, 2023-02-07:
> Hi, try to find openssl.pc on your host machine. If it's missing, probably the package needs to be installed. Then if the compiler still cannot resolve it, add the dir containing the .pc file to the PKG_CONFIG_PATH.

I also have the same problem. I use "export PKG_CONFIG_PATH="/usr/local/opt/openssl/lib/pkgconfig" in terminal. it's no problem, but what need i do in xcode for this problem?