project-chip/connectedhomeip #17239: Unable to cross-compile for 32 bit arm

stlevkov, 2022-04-11:
#### Problem
The project was successfully compiled for bridge and chip-tool using this commit:
commit [c2be9bcece2a7528960628f267cf9132ee7ac5fb]
   - No compilation errors

But after switching to a new:
commit [e92157a98e112326364a8475dd50324f4bbcf1d9]
   - Error compiling

 * expected behavior
No errors during compilation

 * actual behavior
error: format ‘%ld’ expects argument of type ‘long int’, but argument 4 has type ‘size_t’ {aka ‘unsigned int’

 * system configuration
Architecture: armv7l Cortex-A8

Log:
`../../third_party/connectedhomeip/src/platform/Linux/DeviceInfoProviderImpl.cpp:62:68: note: format string is defined here
 
62 | ChipLogProgress(DeviceLayer, "Get the fixed label with index:%ld at endpoint:%d", mIndex, mEndpoint);`

 #### Proposed Solution
use "%zd", instead of %ld, for site_t of mIndex argument

stlevkov, 2022-09-19:
Hi,
Cannot be reproduce with recent commits. Tested with [a8d12af4be0cbf26f3f1352a5562eb7856ca8651]

closed 2022-09-19 by stlevkov

vindicatorr, 2023-01-02:
@stlevkov, I'm experiencing this issue as well.

```
connectedhomeip/config/esp32/third_party/connectedhomeip/src/app/util/mock/attribute-storage.cpp:281:35: error: format '%lx' expects argument of type 'long unsigned int', but argument 4 has type 'int' [-Werror=format=]
[build]   281 |     ChipLogDetail(DataManagement, "Reading Mock Endpoint %" PRIx32 "Mock Cluster %" PRIx32 ", Field %" PRIx32 " is dirty",
[build]       |                                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[build]   282 |                   aPath.mEndpointId, aPath.mClusterId, aPath.mAttributeId);
[build]       |                   ~~~~~~~~~~~~~~~~~
[build]       |                         |
[build]       |                         int
```

I tried looking at the build logs that you were passing, but I wasn't seeing which toolchains were being used.
I was looking for `-esp-elf-g++` in the workflow build logs, and wasn't finding anything. I guess the details are hidden from the ninja output.

I thought I'd make fixes myself and issue my first-ever PR, but after following the "Contributing Guidelines", and running the `Builds` workflow (manually since I missed out on enabling it prior to subsmission), my PR changes failed the workflow check.

I then came across this lengthy thread https://github.com/espressif/esp-idf/issues/9511 which refers to https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/migration-guides/release-5.x/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler.
Note how the fix looks to only be for xtensa, and riscv was already correct. So I'd guess if your commits were passing, that they were only being built for xtensa, and not riscv, and/or not using the latest esp-idf?
I always try to build the latest of everything when I can.

I'll admit I built my own toolchain for my RISC-V esp32-c3, but for due-diligence, I went ahead and downloaded their release of `riscv32-esp-elf-gcc11_2_0-esp-2022r1-linux-amd64.tar.xz` from https://github.com/espressif/crosstool-NG/releases, and that too failed.
My toolchain build was done in November, theirs was July.