project-chip/connectedhomeip #17271: Modeselect cluster [TC-MOD-3.3] - Expected outcome is not matched with the result

somu1710, 2022-04-12:
TC-MOD-3.3 - The commands which are executed using OTA-Requestor app it shows "unsupported cluster" for endpoint "0" and "unsupported endpoint" for endpoint "1".

 ./chip-tool modeselect read start-up-mode 321 0
  { Response Failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)}

./chip-tool modeselect read start-up-mode 321 1
{Response Failure: IM Error 0x0000057F: General error: 0x7f (UNSUPPORTED_ENDPOINT)}

[Mod t3.3 fail.txt](https://github.com/project-chip/connectedhomeip/files/8469629/Mod.t3.3.fail.txt)

[mod 3.3 f.txt](https://github.com/project-chip/connectedhomeip/files/8469627/mod.3.3.f.txt)

woody-apple, 2022-05-19:
Cert Blocker Review: Also assigning @bzbarsky-apple for clarifications

bzbarsky-apple, 2022-05-19:
Nothing requires any particular application to implement the Mode Select cluster.  If the endpoint does not claim conformance to the Mode Select device type, or some other device type that uses that cluster (none of which are defined so far) then it will not have that cluster.

bzbarsky-apple, 2022-05-19:
@cjandhyala What needs to be fixed on the certification side here so we are not trying to require this cluster to exist when it doesn't have to?

closed 2022-05-19 by bzbarsky-apple