project-chip/connectedhomeip #17272: Network Commissioning - ConnectNetwork command is not working when SSID is in Networks attribute list

Sivaramgrl, 2022-04-12:
#### Problem
Networks attribute has one entry with Connect state as FALSE but InterfaceEnabled attribute value is TRUE the issue is followed in [(#17106](https://github.com/project-chip/connectedhomeip/issues/17106)). So using the ConnectNetwork command or remove command or ReorderNetwork command the Chip-tool responded with NetworkingStatus as 3 ( NetworkIDNotFound ) for the SSID value which is listed in the Networks attribute.

###  expected behavior
The commands should be responded with NetworkingStatus as 0 when the connected SSID value is removed or re-ordered.
###  actual behavior
The commands were responded with NetworkingStatus as 3 ( NetworkIDNotFound) when the InterfaceEnabled attribute value is TRUE

###  logs
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning read networks 1 0
[1649742717.698750][2498:2503] CHIP:TOO:   Networks: 1 entries
[1649742717.698901][2498:2503] CHIP:TOO:     [1]: {
[1649742717.698943][2498:2503] CHIP:TOO:       NetworkID: 47524C507269766174655F455854
[1649742717.698980][2498:2503] CHIP:TOO:       **Connected: FALSE**
[1649742717.699017][2498:2503] CHIP:TOO:      }
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning read interface-enabled  1 0
[1649742732.789560][2507:2512] CHIP:TOO:   **InterfaceEnabled: TRUE**
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning reorder-network 47524C507269766174655F455854 1 0 1 0
[1649742844.867710][2516:2521] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1649742844.867804][2516:2521] CHIP:TOO:   NetworkConfigResponse: {
[1649742844.867842][2516:2521] CHIP:TOO:     **networkingStatus: 3**
[1649742844.867881][2516:2521] CHIP:TOO:    }
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning connect-network 47524C507269766174655F455854 0 1 0
[1649742879.569360][2525:2530] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0007
[1649742879.569465][2525:2530] CHIP:TOO:   ConnectNetworkResponse: {
[1649742879.569501][2525:2530] CHIP:TOO:     **networkingStatus: 3**
[1649742879.569532][2525:2530] CHIP:TOO:     debugText:
[1649742879.569563][2525:2530] CHIP:TOO:     errorValue: 0
[1649742879.569594][2525:2530] CHIP:TOO:    
ubuntu@ubuntu:~/sethu-apr1/connectedhomeip/examples/chip-tool/out/debug$ ./chip-tool networkcommissioning remove-network 47524C507269766174655F455854 0 1 0
[1649742897.180070][2532:2537] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1649742897.180148][2532:2537] CHIP:TOO:   NetworkConfigResponse: {
[1649742897.180180][2532:2537] CHIP:TOO:     **networkingStatus: 3**
[1649742897.180212][2532:2537] CHIP:TOO:    }

[networks_issue.txt](https://github.com/project-chip/connectedhomeip/files/8469962/networks_issue.txt)

erjiaqing, 2022-04-13:
Seems you are not using the correct notation, please try

```
./chip-tool networkcommissioning connect-network hex:47524C507269766174655F455854 0 1 0
```

Note the hex prefix

Sivaramgrl, 2022-04-19:
Thanks for sharing the input @erjiaqing,
Based on your comments, I worked on the Networks commissioning cluster commands
Below is my observation:

### Used Commit id:
commit https://github.com/project-chip/connectedhomeip/commit/87938704ed93dde93e960605ae6a714923efa2e0
Date: Mon Apr 18 18:59:53 2022 -0700

### Steps followed:
before the failsafe timeout, I send the add or update network request message.

### Output:
As the sequence is executed before failsafe timeout, I was able to get the response as expected.
Confirmed the output by reading the Networks attribute. The list holds the SSID which is requested in add or update Network command. but still, the list is updated as the connected status is FALSE. Which is tracked in different bugs https://github.com/project-chip/connectedhomeip/issues/17106

Logs:
[Full DUT report-NC.txt](https://github.com/project-chip/connectedhomeip/files/8511615/Full.DUT.report-NC.txt)
[Full TH report -NC.txt](https://github.com/project-chip/connectedhomeip/files/8511623/Full.TH.report.-NC.txt)

As the issue is resolved now. I am closing this ticket.

closed 2022-04-19 by Sivaramgrl