project-chip/connectedhomeip #17289: the new fabric event is possible to collide with the previous removed fabric index

yunhanw-google, 2022-04-12:
#### Problem
Currently All Event have been logged in circular event buffer. When fabric is removed, the already logged event would not be removed until buffer is full(when it is full, the oldest one would be removed). When read/subscribe comes, server would try to use secure session(which has fabric info) to look up events in buffer via comparing with the fabric inside AccessControl::Check, if fabric is not right, then access denies error is generated and would be packed into report.

According to spec, those events have to disappear after the fabric is removed
Otherwise if enough fabrics are re-added, we would have events that are colliding the previous FabricIndex

yunhanw-google, 2022-04-25:
current idea is to record persistent event number for each fabric in fabric table, when new fabric is created, access control can snapshot last event number. When logEvent with fabric index is called, we encode current event into tlv format along with the current fabric's snapshotted event number in circular event tlv buffer . When read handler tries to retrieve fabric events from circular event tlv buffer and deliver report, it would check current readhandler's fabric index and snapshotted event number against the stored one in circular event tlv buffer via access control module.

the 2nd idea is to use timestamp instead of event number correlate with each fabric index for the above.

the third idea is that when the fabric is removed, it call EventManagement to mark the corresponding tlv-encoded fabic-scoped event's fabric index as 0 via scanning all events, we would use TLVUpdater function to mark that fabric index as 0

yunhanw-google, 2022-04-25:
@mrjerryjohns  based upon the conversation with @bzbarsky-apple, the third idea is better one, since the fabric index 0 mean invalid, we just need to locate the related fabric-scope event and update it with invalid 0.

mlepage-google, 2022-04-26:
Something like that is preferable to having access control do something special for this case.

yunhanw-google, 2022-04-28:
https://github.com/project-chip/connectedhomeip/blob/master/src/lib/core/CHIPTLVUpdater.cpp#L59 CHIPTLVUpdater needs the free space so that internally it can use memmove and internal tlvreader and tlvwriter to delete/append new things
currently in circular event tlv buffer, there is no additional free space between event 1 and event 2 and event 3, it seems it is pretty hard to do surgery to update particular tlv fabric index value to 0

I think only first idea to store event number in fabric table works.

bzbarsky-apple, 2022-04-29:
> it seems it is pretty hard to do surgery to update particular tlv fabric index value to 0.

Why?  We should be able to just walk the TLV events and then modify the right bit of memory...

Or honestly, even more simply, walk the TLV and memmove things around to drop all the events for the old fabric, which should be even simpler than updating them...

> it seems the best way is to put event number inside fabric table

It's extra bloat, and we are already hitting RAM limits...  Let's not do this, please.

yunhanw-google, 2022-05-02:
Thanks. Just see the latest comment... in addition, I have created the draft solution for event number inside fabric table for solution reference/discussion purpose last week. 

I think if we use memmove to drop those events, we also need to update the metadata for TLVReader, still need to think about it.

yunhanw-google, 2022-05-03:
after reading through TLVReader, I think I can locate fabric's tag in Event TLV.
    ReturnErrorOnFailure(reader.EnterContainer(containerType));
    ReturnErrorOnFailure(reader.Next(Tag, TLV::ContextTag(1)));
    *(reader.GetReadPoint()) = 0
    reader.ExitContainer(containerType);
So i just need to locate the related fabric-scope event and update it with invalid 0.

mrjerryjohns, 2022-05-04:
I initially recommended the event-number-in-table solution to reduce the complexity. If doing mem-moves of the event buffer is equivalent or lower in complexity, sure, we can do that. Just didn't think that was the case...

mrjerryjohns, 2022-05-04:
Since @yunhanw-google already has a solution to use the event-number based approach, can we get that merged and if needed, do a better solution in the future?

The cost here isn't all that high FYI - it's 8-bytes per fabric for RAM.

closed 2022-05-07 by yunhanw-google