project-chip/connectedhomeip #17310: [SQA] TC-DL-1.1 Cluster revision, Generated command list, Accepted command and attribute list mismatches from test plan.

Survensa, 2022-04-13:
**Issue** : Cluster revision, Generated command list, Accepted command and attribute list mismatches from test plan. in doorlock.

**Expected outcome** :  
- ClusterRevision: 6
-  GeneratedCommandList: 13 entries
- Accepted command list: 32 entries
- Attribute list : 46 entries

**Actual outcome** : 
- ClusterRevision: 3
-  GeneratedCommandList: 0 entries
- AccepctedCommand list: 14 entries
- Attribute list : 29 entries (Including global attributes)
                        
**Commit used** :  fa5561c23d8c1f9464da8669931affbae0660f51

**Log** : 
Command used for Cluster Revision :  `./chip-tool doorlock read cluster-revision 1 1`

> [1649840958.059458][11869:11874] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_FFFD DataVersion: 4187210024
[1649840958.059552][11869:11874] CHIP:TOO:   ClusterRevision: 3
[1649840958.059643][11869:11874] CHIP:EM: Sending Standalone Ack for MessageCounter:328996 on exchange 42344i

Command used for generated command list :  ` ./chip-tool doorlock read generated-command-list 1 1`

> [1649841171.062560][11886:11891] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_FFF8 DataVersion: 4187210024
[1649841171.062650][11886:11891] CHIP:TOO:   GeneratedCommandList: 0 entries
[1649841171.062758][11886:11891] CHIP:EM: Sending Standalone Ack for MessageCounter:11901987 on exchange 35309i

Command used to read accepted command list : ` ./chip-tool doorlock read accepted-command-list 1 1`

> [1649841619.994396][11915:11920] CHIP:TOO:   AcceptedCommandList: 14 entries
[1649841619.994436][11915:11920] CHIP:TOO:     [1]: 0
[1649841619.994466][11915:11920] CHIP:TOO:     [2]: 1
[1649841619.994495][11915:11920] CHIP:TOO:     [3]: 11
[1649841619.994524][11915:11920] CHIP:TOO:     [4]: 12
[1649841619.994552][11915:11920] CHIP:TOO:     [5]: 13
[1649841619.994580][11915:11920] CHIP:TOO:     [6]: 14
[1649841619.994608][11915:11920] CHIP:TOO:     [7]: 15
[1649841619.994636][11915:11920] CHIP:TOO:     [8]: 16
[1649841619.994664][11915:11920] CHIP:TOO:     [9]: 26
[1649841619.994693][11915:11920] CHIP:TOO:     [10]: 27
[1649841619.994722][11915:11920] CHIP:TOO:     [11]: 29
[1649841619.994750][11915:11920] CHIP:TOO:     [12]: 34
[1649841619.994779][11915:11920] CHIP:TOO:     [13]: 36
[1649841619.994807][11915:11920] CHIP:TOO:     [14]: 38
[1649841619.994959][11915:11920] CHIP:EM: Sending Standalone Ack for MessageCounter:1471971 on exchange 10798i

Command used to read attribute list: `./chip-tool doorlock read attribute-list 1 1`

> [1649841950.210293][11938:11943] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_FFFB DataVersion: 4187210024
[1649841950.210406][11938:11943] CHIP:TOO:   AttributeList: 29 entries
[1649841950.210440][11938:11943] CHIP:TOO:     [1]: 0
[1649841950.210465][11938:11943] CHIP:TOO:     [2]: 1
[1649841950.210490][11938:11943] CHIP:TOO:     [3]: 2
[1649841950.210515][11938:11943] CHIP:TOO:     [4]: 3
[1649841950.210540][11938:11943] CHIP:TOO:     [5]: 17
[1649841950.210565][11938:11943] CHIP:TOO:     [6]: 18
[1649841950.210589][11938:11943] CHIP:TOO:     [7]: 19
[1649841950.210613][11938:11943] CHIP:TOO:     [8]: 20
[1649841950.210638][11938:11943] CHIP:TOO:     [9]: 21
[1649841950.210662][11938:11943] CHIP:TOO:     [10]: 23
[1649841950.210686][11938:11943] CHIP:TOO:     [11]: 24
[1649841950.210711][11938:11943] CHIP:TOO:     [12]: 25
[1649841950.210735][11938:11943] CHIP:TOO:     [13]: 26
[1649841950.210759][11938:11943] CHIP:TOO:     [14]: 27
[1649841950.210784][11938:11943] CHIP:TOO:     [15]: 33
[1649841950.210808][11938:11943] CHIP:TOO:     [16]: 35
[1649841950.210832][11938:11943] CHIP:TOO:     [17]: 36
[1649841950.210856][11938:11943] CHIP:TOO:     [18]: 37
[1649841950.210881][11938:11943] CHIP:TOO:     [19]: 38
[1649841950.210905][11938:11943] CHIP:TOO:     [20]: 41
[1649841950.210929][11938:11943] CHIP:TOO:     [21]: 43
[1649841950.210953][11938:11943] CHIP:TOO:     [22]: 48
[1649841950.210978][11938:11943] CHIP:TOO:     [23]: 49
[1649841950.211002][11938:11943] CHIP:TOO:     [24]: 51
[1649841950.211026][11938:11943] CHIP:TOO:     [25]: 65528
[1649841950.211051][11938:11943] CHIP:TOO:     [26]: 65529
[1649841950.211076][11938:11943] CHIP:TOO:     [27]: 65531
[1649841950.211100][11938:11943] CHIP:TOO:     [28]: 65532
[1649841950.211125][11938:11943] CHIP:TOO:     [29]: 65533
[1649841950.211319][11938:11943] CHIP:EM: Sending Standalone Ack for MessageCounter:10393443 on exchange 49560i

tehampson, 2022-05-18:
@Survensa just to confirm this is with the all-clusters app right? I am going to continue investigating with this assumption, but it would be good just to get confirmation. Thanks!

tehampson, 2022-05-24:
End of last week I was updating test case spec to reflect the actual values you should be expecting, but someone beat me to it https://github.com/CHIP-Specifications/chip-test-plans/pull/1270. That PR updates the TC to reflect what values you should expect.

Running this on the all cluster app today I see feature map of:
```
[1653409461.239317][1088162:1088167] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0101 Attribute 0x0000_FFFC DataVersion: 2331464980
[1653409461.239354][1088162:1088167] CHIP:TOO:   FeatureMap: 275
```

GeneratedCommandList:
```
[1653409418.682746][1088102:1088107] CHIP:TOO:   GeneratedCommandList: 5 entries
[1653409418.682827][1088102:1088107] CHIP:TOO:     [1]: 12
[1653409418.682897][1088102:1088107] CHIP:TOO:     [2]: 15
[1653409418.682911][1088102:1088107] CHIP:TOO:     [3]: 28
[1653409418.682924][1088102:1088107] CHIP:TOO:     [4]: 35
[1653409418.682935][1088102:1088107] CHIP:TOO:     [5]: 37
```

AcceptedCommandList
```
[1653409447.162688][1088141:1088146] CHIP:TOO:   AcceptedCommandList: 14 entries
[1653409447.162706][1088141:1088146] CHIP:TOO:     [1]: 0
[1653409447.162718][1088141:1088146] CHIP:TOO:     [2]: 1
[1653409447.162727][1088141:1088146] CHIP:TOO:     [3]: 3
[1653409447.162736][1088141:1088146] CHIP:TOO:     [4]: 11
[1653409447.162745][1088141:1088146] CHIP:TOO:     [5]: 12
[1653409447.162757][1088141:1088146] CHIP:TOO:     [6]: 13
[1653409447.162767][1088141:1088146] CHIP:TOO:     [7]: 14
[1653409447.162778][1088141:1088146] CHIP:TOO:     [8]: 15
[1653409447.162787][1088141:1088146] CHIP:TOO:     [9]: 26
[1653409447.162796][1088141:1088146] CHIP:TOO:     [10]: 27
[1653409447.162806][1088141:1088146] CHIP:TOO:     [11]: 29
[1653409447.162811][1088141:1088146] CHIP:TOO:     [12]: 34
[1653409447.162820][1088141:1088146] CHIP:TOO:     [13]: 36
[1653409447.162829][1088141:1088146] CHIP:TOO:     [14]: 38
```

--------------------------------------------

From accepted command list, seems like 16, 17, 18, 19 are missing 

From generated command list, seems like 18 is still missing.

From attribute list, seems 0xFFFA (65530) is missing

Survensa, 2022-05-26:
> @Survensa just to confirm this is with the all-clusters app right? I am going to continue investigating with this assumption, but it would be good just to get confirmation. Thanks!

We use lock-app for Door lock execution.

tehampson, 2022-05-26:
>We use lock-app for Door lock execution.

With https://github.com/project-chip/connectedhomeip/pull/18711 now in this issue should be resolved

closed 2022-05-26 by tehampson