project-chip/connectedhomeip #17312: [TC-IDM-3.2]--In test step 25 "DUT Fails to send Write Response message with the error DATA_VERSION_MISMATCH"

Rajashreekalmane, 2022-04-13:
#### Problem
As per Test-Plan [TC-IDM-3.2] test-step 25 expected that DUT should sends a Write Response message with the error DATA_VERSION_MISMATCH for the second Write request, when TH sends a second Write 
Request Message to the DUT to modify the value of an attribute with the dataversion field set to
 the value received earlier.
 
* expected behavior:-DATA_VERSION_MISMATCH(0x92)
 * actual behavior:-SUCCESS (0x00)

 * steps to reproduce:-
 1. ./chip-tool onoff read start-up-on-off 1 1
 2.  ./chip-tool onoff write start-up-on-off 1 1 1
 3. ./chip-tool onoff write start-up-on-off 2 1 1 --data-version 0x365b6fd5

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#write-response-action

Test-Plan screenshot:-
[image]

PFA log below:-
[TC-IDM-3.2_step 25.txt](https://github.com/project-chip/connectedhomeip/files/8480198/TC-IDM-3.2_step.25.txt)

vivien-apple, 2022-04-20:
@yunhanw-google Do you have any idea of what is going on ?

yufengwangca, 2022-04-25:
I saw following error as I try to repo this issue.

```
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ ./chip-tool onoff write start-up-on-off 2 12344321 1 --data-version 0x1172457513
[1650930536.603139][1341278:1341278] CHIP:TOO: InitArgs: Invalid argument data-version: 0x1172457513
Usage:
  ./chip-tool onoff write attr-name attr-value node-id/group-id endpoint-id-ignored-for-group-commands [--paa-trust-store-path] [--commissioner-name] [--commissioner-nodeid] [--commissioner-fabricid] [--trace_file] [--trace_log] [--ble-adapter] [--timedInteractionTimeoutMs] [--data-version] [--suppressResponse] [--timeout]
[1650930536.603190][1341278:1341278] CHIP:TOO: Run command failure: ../../../examples/chip-tool/commands/common/Commands.cpp:137: Error 0x0000002F
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ 
yufengwang@yufengwang:~/connectedhomeip/out/debug/standalone$ 
```

yunhanw-google, 2022-04-26:
From the log, it should the last WriteRequest don't have dataVersion inside, it should have dataVersion inside, in addition, this logic has been validated in python, so there is the regression from chip-tool's WriteAttribute dataVersion to WriteClient::EncodeAttribute. @erjiaqing  could you take a look at this?

yunhanw-google, 2022-04-27:
https://github.com/project-chip/connectedhomeip/pull/17801

closed 2022-04-27 by yunhanw-google