project-chip/connectedhomeip #17319: [TC-GC-2.4] While verifying SetRegulatoryConfig command we will get Success value(0) instead 'Error - ValueOutsideRange' value(1)

Rajashreekalmane, 2022-04-13:
#### Problem
As per Test-Plan, test-step 9 expect that " SetRegulatoryConfigResponse command response from DUT to TH1 with field ErrorCode as 'Error - ValueOutsideRange' and DebugText as a 'string' value" when TH1 sends SetRegulatoryConfig command to the DUT with NewRegulatoryConfig data value as '3', Breadcrumb as 0 and CountryCode is of type 'string'. But we are getting response as success with the value 0
 
* expected behavior:- Th should send error code as ValueOutsideRange(1)
 * actual behavior:-TH responds with error code success(0)
 * steps to reproduce:-./chip-tool generalcommissioning set-regulatory-config 3 te9 0 1 0

Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#tc-gc-2-4-verification-for-errorcode-message-dut-commissionee

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/GeneralCommissioningCluster.adoc#ref_CommissioningError
 
PFA log below:-
[Log.txt](https://github.com/project-chip/connectedhomeip/files/8481283/Log.txt)

andy31415, 2022-04-25:
@Rajashreekalmane what device was this tested on?

Survensa, 2022-04-26:
> @Rajashreekalmane what device was this tested on?

Its was tested using both RPI (ble-wifi) and Thread .

PR #18067 (2022-05-05): Implement rules for SetRegulatoryConfig command
#### Problem
What is being fixed?  Examples:
* As per Test-Plan, test-step 9 expect that " SetRegulatoryConfigResponse command response from DUT to TH1 with field ErrorCode as 'Error - ValueOutsideRange' and DebugText as a 'string' value" when TH1 sends SetRegulatoryConfig command to the DUT with NewRegulatoryConfig data value as '3', Breadcrumb as 0 and CountryCode is of type 'string'. But we are getting response as success with the value 0

* Fixes #17319

#### Change overview
Implement rules for SetRegulatoryConfig command

#### Testing
How wa [... cut]
closed 2022-05-05 by yufengwangca