project-chip/connectedhomeip #17327: Added one variable in config section but getting error when executed

manjunathambiger23, 2022-04-13:
Summary:
               [ TC-DM-2.2 ] tried adding supportedFabric value as 16 in config section getting zap error when executed: TypeError: value.replace is not a function, below i added YAML and execution screenshot.

YAML screenshot: 
[image]
[image]

Execution logs and screenshot:
[image]
[Execution_log.txt](https://github.com/project-chip/connectedhomeip/files/8482028/Execution_log.txt)

bzbarsky-apple, 2022-04-22:
I just tried this on top of Matter revision 6109d14348023229614227775b57b8b4c80f131a with the following diff applied:
```
diff --git a/src/app/tests/suites/certification/Test_TC_DM_2_2.yaml b/src/app/tests/suites/certification/Test_TC_DM_2_2.yaml
index 625417b1f..c7024ecee 100644
--- a/src/app/tests/suites/certification/Test_TC_DM_2_2.yaml
+++ b/src/app/tests/suites/certification/Test_TC_DM_2_2.yaml
@@ -18,6 +18,9 @@ config:
     nodeId: 0x12344321
     cluster: "Operational Credentials"
     endpoint: 0
+    supportFabric:
+        type: INT8U
+        defaultValue: 16
 
 tests:
     - label: "Wait for the commissioned device to be retrieved"
@@ -69,7 +72,7 @@ tests:
       command: "readAttribute"
       attribute: "SupportedFabrics"
       response:
-          value: 16
+          value: supportFabric
           constraints:
               type: uint8
 
```

and things generated fine.  @manjunathambiger23 what are the exact steps to reproduce this?  Please attach a diff to apply as needed, or better yet please link to a branch in github with the relevant changes.

bzbarsky-apple, 2022-05-22:
No response, can't reproduce, resolving. 

@cjandhyala

closed 2022-05-22 by bzbarsky-apple