project-chip/connectedhomeip #17351: GetAttestationChallenge is a public API and doesn't make sense

cecille, 2022-04-13:
#### Problem
GetAttestationChallenge ended up as a public API. BUT, it really is just a convenience for the commissioner to get the attestation challenge of the device CURRENTLY being commissioned because the proxy isn't provided. Previously, this (accidentally IMO) fell back to the most recently PASE'd proxy, but as we moved towards less statefulness, this function got left behind.

 #### Proposed Solution
Add optional var to the API to allow a proxy to be provided, THEN fall back to the current commissioning device.
CHIP_ERROR GetAttestationChallenge(ByteSpan & attestationChallenge, const DeviceProxy* proxy = nullptr)

bzbarsky-apple, 2022-04-22:
Alternately, how about we just move the API to `DeviceProxy`, since it does not actually involve any commissioner state?

cecille, 2022-04-25:
The reason I suggested the default parameter was in case devs are using it already. It does make more sense as part of the device proxy, but post-FC, I'm not sure such an API chance would be allowed.

bzbarsky-apple, 2022-04-25:
My point was that we should add an API on DeviceProxy.  The API on commissioner can remain, and call the DeviceProxy API on the commmissionnee device, to preserve existing behavior.

This will not break any existing consumers, as far as I can tell.