project-chip/connectedhomeip #17380: Level control min and max level issue

pankore, 2022-04-14:
#### Problem
In level-control.cpp, the min and max levels are defined as shown below
`#define LEVEL_CONTROL_LIGHTING_MIN_LEVEL 0x01`
`#define LEVEL_CONTROL_LIGHTING_MAX_LEVEL 0xFE`

However, chip-tool allows level to be 0 and 255
`./chip-tool levelcontrol move-to-level-with-on-off 255 0 1234 1`
`./chip-tool levelcontrol move-to-level-with-on-off 0 0 1234 1`

bzbarsky-apple, 2022-05-22:
chip-tool allows sending invalid values of all sorts; it's a test tool.

The cluster impl should be checking for a valid range and failing out, and it sounds like it's not doing that.

@jmartinez-silabs

jmartinez-silabs, 2022-05-22:
For a lighting device the minimal is 1 but if it is not a lighting device aka featuremap not at 1 you can go as low as 0. I am 99% sure those range are checked and applied. It is even tested in one of the yaml test case

bzbarsky-apple, 2022-05-23:
@jmartinez-silabs Yes, the check for 255 is what is missing.... Per spec, that should error out, not silently treat the input as 254.

closed 2022-05-31 by andy31415