project-chip/connectedhomeip #17390: Commissioning doesnot work with chip-tool

PSONALl, 2022-04-14:
#### Problem
Commissioning and cluster control does not work on my mac
chip-tool gives timeout during commissioning
> [1649926750459] [50648:3211908] CHIP: [TOO] Run command failure: ../../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
 * system configuration 
 - Using macOS BigSur version 11.6 
 - DUT: esp32 app: all-clusters-app
 
 * logs
 - esp32 logs: [esp32 .txt](https://github.com/project-chip/connectedhomeip/files/8487970/esp32.txt)
- DUT logs: [chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8487974/chip-tool.txt)

 #### Proposed Solution
- I think this is an issue specific to the macOS platform
- Commissioning with Linux chip-tool works fine

tcarmelveilleux, 2022-04-14:
It appears that the ESP32 was not found over mDNS after 10 seconds
```
[1649926819312] [50678:3214275] CHIP: [CSM] FindOrEstablishSession: PeerId = 577E28D4891C4D0A:0000000000000001
[1649926819312] [50678:3214275] CHIP: [CSM] FindOrEstablishSession: No existing session found
[1649926819312] [50678:3214275] CHIP: [DIS] Resolving 577E28D4891C4D0A:0000000000000001 ...
[1649926819514] [50678:3214276] CHIP: [DIS] Checking node lookup status after 202 ms
[1649926829315] [50678:3214268] CHIP: [CTL] Shutting down the commissioner
[1649926829315] [50678:3214268] CHIP: [CTL] Shutting down the controller
[1649926829315] [50678:3214268] CHIP: [CTL] Shutting down the commissioner
[1649926829315] [50678:3214268] CHIP: [CTL] Shutting down the controller
[1649926829315] [50678:3214268] CHIP: [DIS] Cancelling incomplete address resolution as device is being deleted.
[1649926829315] [50678:3214268] CHIP: [DIS] Discovery does not require any more timeouts
```

The ESP32 received a ConnectNetwork command, and took a couple seconds to connect to Wi-Fi and start advertising operationally:
```
I (54862) chip[DMG]: AccessControl: implicit admin (PASE)
I (54872) chip[IN]: Prepared secure message 0x3ffe2b44 to 0xFFFFFFFB00000000 (1)  of type 0x9 and protocolId (0, 1) on exchange 58115r with MessageCounter:6040811.
I (54882) chip[IN]: Sending encrypted msg 0x3ffe2b44 with MessageCounter:6040811 to 0xFFFFFFFB00000000 (1) at monotonic time: 000000000000D338 msec
I (54902) NimBLE: GATT procedure initiated: notify; 
I (54902) NimBLE: att_handle=14

I (54912) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (54932) CHIP[DL]: Write request received for CHIPoBLE RX characteristic con 0 12
I (54942) chip[EM]: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:1820146 on exchange 58116r
I (54952) chip[DMG]: AccessControl: checking f=1 a=p s=0xFFFFFFFB00000000 t= c=0x0000_0031 e=0 p=a
I (54952) chip[DMG]: AccessControl: implicit admin (PASE)
I (54972) chip[NP]: ESP NetworkCommissioningDelegate: SSID: ES_Staff
I (54972) chip[DL]: WiFi station mode change: Enabled -> Disabled
I (55002) chip[DL]: Changing ESP WiFi mode: STA -> STA+AP
I (55002) wifi:mode : sta (c4:dd:57:5b:f7:f0) + softAP (50:43:03:ce:12:87)

.... some more status ....

I (56922) chip[DL]: WIFI_EVENT_STA_CONNECTED
I (56922) chip[DL]: WiFi station state change: Connecting -> Connecting_Succeeded
I (56922) chip[DL]: WiFi station state change: Connecting_Succeeded -> Connected
I (56932) chip[DL]: WiFi station interface connected
I (56942) chip[DIS]: Advertise operational node 577E28D4891C4D0A-0000000000000001
I (56952) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (56952) chip[DIS]: mDNS service published: _matter._tcp
E (56962) chip[SVR]: Operational advertising enabled
I (56972) app-devicecallbacks: Current free heap: 89736

I (56982) chip[IN]: Prepared secure message 0x3ffe2d44 to 0xFFFFFFFB00000000 (1)  of type 0x9 and protocolId (0, 1) on exchange 58116r with MessageCounter:6040812.
I (57002) chip[IN]: Sending encrypted msg 0x3ffe2d44 with MessageCounter:6040812 to 0xFFFFFFFB00000000 (1) at monotonic time: 000000000000DB78 msec
I (56992) wifi:AP's beacon interval = 102400 us, DTIM period = 2
I (57012) NimBLE: GATT procedure initiated: notify; 
I (57012) NimBLE: att_handle=14

I (57022) chip[DL]: Confirm received for CHIPoBLE TX characteristic indication (con 0) status= 0 
I (57032) chip[DL]: Done driving station state, nothing else to do...
I (57032) app-devicecallbacks: Current free heap: 88172

E (57042) chip[DL]: Long dispatch time: 126 ms, for event type 49152
I (57052) chip[DL]: Updating advertising data
I (57062) app-devicecallbacks: Current free heap: 89692

I (57522) esp_netif_handlers: sta ip: 192.168.126.58, mask: 255.255.255.0, gw: 192.168.126.49
I (57522) chip[DL]: IP_EVENT_STA_GOT_IP
I (57522) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.126.58/255.255.255.0 gateway 192.168.126.49
I (57532) chip[DL]: IPv4 Internet connectivity ESTABLISHED
I (57552) app-devicecallbacks: Current free heap: 89708

I (57552) app-devicecallbacks: Server ready at: 192.168.126.58:5540
I (57562) chip[DIS]: Updating services using commissioning mode 0
I (57562) chip[DIS]: CHIP minimal mDNS started advertising.
I (57572) chip[DIS]: Advertise operational node 577E28D4891C4D0A-0000000000000001
I (57572) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (57592) chip[DIS]: mDNS service published: _matter._tcp
[... 24 lines cut ...]
I (60022) chip[DL]: IP_EVENT_GOT_IP6
I (60022) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: 2409:4042:4e99:6dcc:c6dd:57ff:fe5b:f7f0
I (60032) chip[DL]: IPv6 Internet connectivity ESTABLISHED
I (60032) app-devicecallbacks: Current free heap: 89148

I (60032) app-devicecallbacks: IPv6 Server ready...
I (60052) chip[DIS]: Updating services using commissioning mode 0
I (60052) chip[DIS]: CHIP minimal mDNS started advertising.
I (60072) chip[DIS]: Advertise operational node 577E28D4891C4D0A-0000000000000001
I (60082) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (60092) chip[DIS]: mDNS service published: _matter._tcp
I (60092) app-devicecallbacks: Current free heap: 87348

I (60092) chip[DIS]: Updating services using commissioning mode 0
I (60112) chip[DIS]: CHIP minimal mDNS started advertising.
I (60122) chip[DIS]: Advertise operational node 577E28D4891C4D0A-0000000000000001
I (60132) chip[DIS]: CHIP minimal mDNS configured as 'Operational device'.
I (60152) chip[DIS]: mDNS service published: _matter._tcp
I (60152) ROUTE_HOOK: Hook already installed on netif, skip...
I (60152) app-devicecallbacks: Current free heap: 86348
```

Overall, it took approx 6 seconds before all IP addresses were obtained. It seems that the mac chip-tool did not see the mDNS advertisements.

This appears to be an issue that can only be diagnosed with logs from what the MacOS-side DNS-SD querier was seeing.

bzbarsky-apple, 2022-04-14:
@PSONALl `log stream --line-buffered --system --level=info --filter 'process:"mDNSResponder"'` should help get the DNS-SD logs on the Mac.  might need the mDNSResponder debugging profile at https://developer.apple.com/bug-reporting/profiles-and-logs/?name=dns installed (takes effect immediately on install, no reboot needed) to show all the data we care about.

cecille, 2022-04-14:
We found this same issue earlier on linux where the M5 was taking too long to respond. We fixed this by adding retries in the avahi and minimal logic. It would appear that the M5 has now crossed the mac timeout threshold as well and may require retries there.

avahi retry PR: https://github.com/project-chip/connectedhomeip/pull/12923

PSONALl, 2022-04-15:
ok thanks @bzbarsky-apple 
here are the logs of mDNSResponder : [dns adv.txt](https://github.com/project-chip/connectedhomeip/files/8496268/dns.adv.txt)

PSONALl, 2022-04-18:
from logs I think MDNS querier can see _matter._tcp.local service,
so what can be the issue here? @bzbarsky-apple @tcarmelveilleux ?

bzbarsky-apple, 2022-04-18:
Are the time stamps in that "dns adv" log correlated with the device logs?  Doesn't seem to me like they are.  It would be useful to have a coherent set of logs from a single comissioning failure attempt.

Looking at the chip-tool log this is the relevant bit, as far as I can tell:
```
[1649926735448] [50648:3211934] CHIP: [CTL] Performing next commissioning step 'FindOperational' with completion status = '../../third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1764: Success'
[1649926735448] [50648:3211934] CHIP: [CSM] FindOrEstablishSession: PeerId = 577E28D4891C4D0A:0000000000000001
[1649926735448] [50648:3211934] CHIP: [CSM] FindOrEstablishSession: No existing session found
[1649926735448] [50648:3211934] CHIP: [DIS] Resolving 577E28D4891C4D0A:0000000000000001 ...
[1649926735452] [50648:3211934] CHIP: [DIS] Operational node lookup already in progress. Will NOT start a new one.
[1649926735452] [50648:3211934] CHIP: [DMG] ICR moving to [AwaitingDe]
[1649926735649] [50648:3211936] CHIP: [DIS] Checking node lookup status after 200 ms
[1649926750449] [50648:3211936] CHIP: [DIS] Checking node lookup status after 15000 ms
[1649926750449] [50648:3211936] CHIP: [DIS] Operational discovery failed for 0x0000000000000001: ../../third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:174: CHIP Error 0x00000032: Timeout
```
So interesting questions:
1. Why "Operational node lookup already in progress."?  Where did that come from?  Are there multiple calls to `OperationalDeviceProxy::LookupPeerAddress` here?  With what callstacks?
2. Why are we not seeing any IP addresses after 15 seconds in the resolution code?  This is where a time-correlated log against what `mDNSResponder` thinks is going on would be helpful.

@PSONALl

PSONALl, 2022-04-20:
Attaching logs again, strange but this time I am getting some different error logs, I have changed nothing using same HEAD:
[dns-responder.txt](https://github.com/project-chip/connectedhomeip/files/8521141/dns-responder-new.txt)
[chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8521143/chip-tool-new.txt)
[esp32.txt](https://github.com/project-chip/connectedhomeip/files/8521144/esp32-new.txt)

bzbarsky-apple, 2022-04-20:
OK.  So now the log shows that commissioning works fine.  The node is commissioned, discovered on the operational network at `192.168.165.58%en0:5540` (because the esp32 brings up this IP almost a second before it brings up IPV6 connectivity), a CASE session is established and we send `CommissioningComplete`, the `pairing ble-wifi` command completes correctly.

Now chip-tool tries to send the "on" command.   It resolves the esp32 fine:
```
[1650449951518] [22708:249318] CHIP: [DIS] Node ID resolved for EEDBFC2E8E4B538F:0000000000000001
[1650449951518] [22708:249318] CHIP: [DIS]     Addr 0: [fe80::c6dd:57ff:fe5b:f7f0]:5540
[1650449951518] [22708:249318] CHIP: [DIS]     Addr 1: [2405:204:949f:2190:c6dd:57ff:fe5b:f7f0]:5540
[1650449951518] [22708:249318] CHIP: [DIS]     Addr 2: [192.168.165.58]:5540
[1650449951519] [22708:249318] CHIP: [DIS] UDP:[fe80::c6dd:57ff:fe5b:f7f0%en0]:5540: new best score: 3
[1650449951519] [22708:249318] CHIP: [DIS] Lookup clearing interface for non LL address
[1650449951519] [22708:249318] CHIP: [DIS] UDP:[2405:204:949f:2190:c6dd:57ff:fe5b:f7f0%en0]:5540: new best score: 5
[1650449951519] [22708:249318] CHIP: [DIS] UDP:192.168.165.58%en0:5540: score has not improved: 2
[1650449951519] [22708:249318] CHIP: [DIS] Checking node lookup status after 207 ms
[1650449951519] [22708:249318] CHIP: [CTL] Updating device address to UDP:[2405:204:949f:2190:c6dd:57ff:fe5b:f7f0]:5540 while in state 1
[1650449951519] [22708:249318] CHIP: [CTL] OperationalDeviceProxy[EEDBFC2E8E4B538F:0000000000000001]: State change 1 --> 2
[1650449951520] [22708:249318] CHIP: [SC] Establishing CASE session to UDP:[2405:204:949f:2190:c6dd:57ff:fe5b:f7f0]:5540
```
but fails to send Sigma1 to it at that IP:
```
[1650449951523] [22708:249318] CHIP: [-] ../../third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:417: OS Error 0x02000041: No route to host at ../../commands/clusters/ModelCommand.cpp:53
```

The log from the ESP32 does show it using those IPs:
```
I (14517) chip[DL]: IPv4 address changed on WiFi station interface: 192.168.165.58/255.255.255.0 gateway 192.168.165.161
...
I (16467) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: fe80:0000:0000:0000:c6dd:57ff:fe5b:f7f0
...
I (16497) chip[DL]: IPv6 addr available. Ready on WIFI_STA_DEF interface: 2405:0204:949f:2190:c6dd:57ff:fe5b:f7f0
```

@PSONALI a good first thing to check is whether you can `ping6 2405:204:949f:2190:c6dd:57ff:fe5b:f7f0%en0` (or whatever IP the esp32 is claiming if you have rebooted it since those logs were captured) from your mac.

closed 2023-03-18 by stale[bot]