project-chip/connectedhomeip #17392: [SQA] Door Lock - Missing commands has to be implemented.

Survensa, 2022-04-14:
**Issue** : Door lock cluster has to be implemenated with below commands based on test plan.

**commit used** : fa5561c23d8c1f9464da8669931affbae0660f51		

**Referance** :  https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/door_lock_Cluster.adoc#2-test-case-list

**List of missing command based on the test cases** :

- [TC-DL-2.4] - Unlock with Timeout command
- [TC-DL-2.5] - Get Log Record, Get Log Record Response Command
- [TC-DL-2.6] - Set PIN Code, Get PIN Code, Get PIN Code Response, Clear PIN Code, Clear All PIN Codes
- [TC-DL-2.7] - Set User Status, Get User Status , Get User Status Response
- [TC-DL-2.9] - Set Holiday Schedule, Get Holiday Schedule, Get Holiday Schedule Response , Clear Holiday Schedule
- [TC-DL-2.10]- Set User Type, Get User Type ,Get User Type Response
- [TC-DL-2.12]- Set RFID Code, Get RFID Code, Get RFID Code Response, Clear RFID Code and Clear All RFID Codes
- [TC-DL-2.15]- Operation Event Notification Command
- [TC-DL-2.16]- Programming Event Notification Command

jrhees-cae, 2022-04-16:
Wait a minute... At the very least the Set/Clear pin, etc. and the set/get user status test cases should not be part of a Matter test case. All of these command sets are for !USR pics (e.g. are for ZigBee only)

sethunk, 2022-04-19:
@jrhees-cae , I see that the following command sets are not for !USR 

- Unlock with Timeout command 
- Get Log Record, Get Log Record Response Command
- Set Holiday Schedule, Get Holiday Schedule, Get Holiday Schedule Response , Clear Holiday Schedule
- Operation Event Notification Command
- Programming Event Notification Command

jrhees-cae, 2022-04-19:
That's correct. However please also check other commands, such as get log
record. They are for LOG which I believe is only for ZigBee and not for
Matter.

Jon

jrhees-cae, 2022-04-19:
See the 'Element Requirements' section at
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/device_types/DoorLock.adoc

Jon

Survensa, 2022-04-22:
@jrhees-cae 

After all Zigbee and optional aributes below mentioned command needs to be implemented as it is a SCH feature

- Set Holiday Schedule, Get Holiday Schedule, Get Holiday Schedule Response , Clear Holiday Schedule

closed 2022-05-26 by andy31415