project-chip/connectedhomeip #17405: chip-tool command failed with "Invalid CASE parameters" error

yinyihu-silabs, 2022-04-14:
#### Problem
After commissioning end device over Thread, subsequent chip-tool commands returned "Invalid CASE parameters" error.
**chip-tool logs:**
[1649870251.217524][16138:16143] CHIP:EM: Received message of type 0x40 with protocolId (0, 0) and MessageCounter:2814437257 on exchange 56985i
[1649870251.217624][16138:16143] CHIP:EM: Found matching exchange: 56985i, Delegate: 0xffff940015b0
[1649870251.217713][16138:16143] CHIP:EM: Rxd Ack; Removing MessageCounter:58960980 from Retrans Table on exchange 56985i
[1649870251.217772][16138:16143] CHIP:EM: Removed CHIP MessageCounter:58960980 from RetransTable on exchange 56985i
[1649870251.217867][16138:16143] CHIP:SC: Received error (protocol code 2) during pairing process. ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1479: CHIP Error 0x00000054: Invalid CASE parameter
[1649870251.217926][16138:16143] CHIP:CTL: OperationalDeviceProxy[CBC9D2D911B18741:0000000000002235]: State change 3 --> 2
[1649870251.218024][16138:16143] CHIP:-: ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1479: CHIP Error 0x00000054: Invalid CASE parameter at ../../examples/chip-tool/commands/clusters/ModelCommand.cpp:53
[1649870251.218129][16138:16143] CHIP:EM: Sending Standalone Ack for MessageCounter:2814437257 on exchange 56985i
[1649870251.218215][16138:16143] CHIP:IN: Prepared unauthenticated message 0xffffa8d8e9a8 to 0x0000000000000000 (0)  of type 0x10 and protocolId (0, 0) on exchange 56985i with MessageCounter:58960981.
[1649870251.218291][16138:16143] CHIP:IN: Sending unauthenticated msg 0xffffa8d8e9a8 with MessageCounter:58960981 to 0x0000000000000000 at monotonic time: 00000000009B3243 msec
[1649870251.218551][16138:16143] CHIP:EM: Flushed pending ack for MessageCounter:2814437257 on exchange 56985i
[1649870251.218908][16138:16138] CHIP:CTL: Shutting down the commissioner
[1649870251.219274][16138:16138] CHIP:CTL: Shutting down the controller
[1649870251.219351][16138:16138] CHIP:CTL: Shutting down the commissioner
[1649870251.219511][16138:16138] CHIP:CTL: Shutting down the controller
[1649870251.219678][16138:16138] CHIP:IN: Expiring all connections for fabric 1!!
[1649870251.219771][16138:16138] CHIP:CTL: Shutting down the commissioner
[1649870251.219934][16138:16138] CHIP:CTL: Shutting down the controller
[1649870251.219987][16138:16138] CHIP:IN: Expiring all connections for fabric 2!!
[1649870251.220060][16138:16138] CHIP:CTL: Shutting down the commissioner
[1649870251.220216][16138:16138] CHIP:CTL: Shutting down the controller
[1649870251.220266][16138:16138] CHIP:IN: Expiring all connections for fabric 3!!
[1649870251.220332][16138:16138] CHIP:CTL: Shutting down the System State, this will teardown the CHIP Stack
[1649870251.221198][16138:16138] CHIP:DMG: IM WH moving to [Uninitialized]
[1649870251.221254][16138:16138] CHIP:DMG: IM WH moving to [Uninitialized]
[1649870251.221298][16138:16138] CHIP:DMG: IM WH moving to [Uninitialized]
[1649870251.221341][16138:16138] CHIP:DMG: IM WH moving to [Uninitialized]
[1649870251.221387][16138:16138] CHIP:DMG: All ReadHandler-s are clean, clear GlobalDirtySet
[1649870251.221540][16138:16138] CHIP:BLE: BleConnectionDelegate::CancelConnection is not implemented.
[1649870251.221961][16138:16138] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-IRacOv)
[1649870251.223118][16138:16138] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1649870251.223230][16138:16138] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1649870251.223286][16138:16138] CHIP:DL: Inet Layer shutdown
[1649870251.223333][16138:16138] CHIP:DL: BLE shutdown
[1649870251.223381][16138:16138] CHIP:DL: System Layer shutdown
[1649870251.223584][16138:16138] CHIP:TOO: Run command failure: ../../examples/chip-tool/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1479: CHIP Error 0x00000054: Invalid CASE parameter

**RTT logs:**
<detail> [DMG] Failed to read stored attribute (1, 0x00<error > [DL] OnSrpClientNotification: Operation refused for security reasons
<info  > [DL] _OnPlatformEvent default:  event->Type = 32780
<detail> [DL] OpenThread State Changed (Flags: 0x00000064)
<detail> [DL]    Device Role: ROUTER
<detail> [DL] Thread Attached updating Multicast address
<info  > [SVR] Joining Multicast groups
<info  > [DL] _OnPlatformEvent default:  event->Type = 32769
<detail> [IN] UDP Message Received packet nb : 11 SrcAddr : FD9C:869B:175:1A35:DD9C:7D06:240C:2821[55643] DestAddr : FD9C:869B:175:1A35:D7F6:9796:6F54:6A79[5540] Payload Length 177
<info  > [EM] Received message of type 0x30 with protocolId (0, 0) and MessageCounter:58960980 on exchange 56985r
<detail> [EM] Handling via exchange: 56985r, Delegate: 0x2001a3e4
<info  > [IN] CASE Server received Sigma1 message. Starting handshake. EC 0x2001b7d0
<info  > [DIS] CASE handshake initiated, closing all BLE Connections
<detail> [SC] Waiting for Sigma1 msg
<info  > [IN] CASE Server disabling CASE session setups
<info  > [SC] Received Sigma1 msg
<detail> [SC] Found MRP parameters in the message
<detail> [SC] Peer assigned session key ID 63799
<error > [SC] CASE failed to match destination ID with local fabrics
<detail> [SC] 0xa6, 0x6d, 0x8a, 0x5d, 0x92, 0xc9, 0x5c, 0x08, 
<detail> [SC] 0x68, 0xb0, 0xf6, 0x6a, 0xa1, 0x62, 0xe3, 0x06, 
<detail> [SC] 0x32, 0x2a, 0xe2, 0xf0, 0xed, 0xf5, 0xf5, 0xaf, 
<detail> [SC] 0x65, 0x9d, 0x4e, 0x42, 0x43, 0x08, 0x2d, 0xe2, 
<detail> [SC] Sending status report. Protocol code 2, exchange 56985
<detail> [EM] Piggybacking Ack for MessageCounter:58960980 on exchange: 56985r
<info  > [IN] Prepared unauthenticated message 0x2001b6ec to 0x0000000000000000 (0)  of type 0x40 and protocolId (0, 0) on exchange 56985r with MessageCounter:2814437257.
<info  > [IN] Sending unauthenticated msg 0x2001b6ec with MessageCounter:2814437257 to 0x0000000000000000 at monotonic time: 00000000000172F7 ms<error > [IN] CASE Session establishment failed: Error CHIP:0x00000003
<info  > [IN] CASE Server enabling CASE session setups
<info  > [IN] CASE Server enabling CASE session setups
<error > [EM] OnMessageReceived failed, err = Error CHIP:0x00000003
<detail> [IN] UDP Message Received packet nb : 12 SrcAddr : FD9C:869B:175:1A35:DD9C:7D06:240C:2821[55643] DestAddr : FD9C:869B:175:1A35:D7F6:9796:6F54:6A79[5540] Payload Length 26
<info  > [EM] Received message of type 0x10 with protocolId (0, 0) and MessageCounter:58960981 on exchange 56985r
<detail> [EM] Found matching exchange: 56985r, Delegate: 0
<detail> [EM] Rxd Ack; Removing MessageCounter:2814437257 from Retrans Table on exchange 56985r
<detail> [EM] Removed CHIP MessageCounter:2814437257 from RetransTable on exchange 56985r

bzbarsky-apple, 2022-04-22:
@yinyihu-silabs Please attach the full log from the command that commissioned the device.

yinyihu-silabs, 2022-04-25:
Here is full commissioning and chip-tool read command log
[Invalid CASE paramter commision.log](https://github.com/project-chip/connectedhomeip/files/8557995/Invalid.CASE.paramter.commision.log)

bzbarsky-apple, 2022-04-26:
@yinyihu-silabs Thank you.  That log shows us discovering the same device as we discovered at the end of commissioning, so the question is what changed between then and the follow-up command.  Can you please attach the full server device log, which would show both the commissioning and the following unsuccessful command, and everything in between?  Ideally along with the client logs corresponding to the process, so the timestamps can be matched up to see what's going on.

yinyihu-silabs, 2022-04-26:
_Invalid CASE parameter_ has become intermittent lately. This is one of the older logs when I was able to catch the error but after commissioning completed. Hope it would help.
Typically in my tests, I only send chip-tool commands (read/write/functional).  If I come across the same failure again, I'll make sure capture everything.
[failed chip-tool.txt](https://github.com/project-chip/connectedhomeip/files/8565729/failed.chip-tool.txt)
[failed chip-tool - rtt log.txt](https://github.com/project-chip/connectedhomeip/files/8565730/failed.chip-tool.-.rtt.log.txt)

yinyihu-silabs, 2022-04-26:
This might be similar to what's been mentioned in https://github.com/project-chip/connectedhomeip/issues/17698
https://github.com/project-chip/connectedhomeip/issues/17698#issuecomment-1110132169

bzbarsky-apple, 2022-04-27:
Again, to make sense of this I would like to see a log that shows:

1. Commissioning.
2. All the bits that work fine.
3. The bit that does not work fine.

For example: the log attached above just says for the server: "CASE failed to match destination ID with local fabrics".  Why did it fail?  Maybe the node was not commissioned onto that fabric.  Maybe commissioning did not complete and then the fail-safe expired and wiped out the partially-comissioned fabric.  Maybe someone came along and deleted the fabric.  It's impossible to say from that server log, because it's missing the part where something actually went wrong and the fabric went away.

closed 2022-05-09 by yinyihu-silabs

hiteshmathukiya, 2022-11-03:
I m facing the same ERROR. @yinyihu-silabs can you suggest some solution?

yinyihu-silabs, 2022-11-03:
@hiteshmathukiya we have not seen this happening too often lately.  Last time we saw it there was some sdk issue and has since been resolved.

hiteshmathukiya, 2022-11-04:
@yinyihu-silabs It resolves as we update the matter SDK. 
- It onboarded and send on-off commands successfully for the master branch Matter SDK.