project-chip/connectedhomeip #17406: chip-tool command failed with "Timeout" error

yinyihu-silabs, 2022-04-14:
#### Problem
After commissioning end device over Thread, subsequent chip-tool commands returned "Timeout" error.

<13:48:33.542> Thread.run:[172.16.0.4:23:O]pi@raspi-04:~$ ~/connectedhomeip/out/chip-tool/chip-tool colorcontrol read cluster-revision 5244 1
<13:48:33.544> MatterCli.sendCmd:DEBUG: expect found after 1 attempt(s)
<13:48:33.545> MatterCli.sendCmd:Node<172.16.0.4> issuing cmd<
> expecting<(.+@.+)(.*)$> timeout<120000> attempt <0
<13:48:33.545> MatterCli.sendCmd:Sent <
>, to <172.16.0.4:23>, regex <(.+@.+)(.*)$>, timeout <120000>
<13:48:33.545> MatterCli.sendCmd:[172.16.0.4:23:I]
<13:48:33.553> Thread.run:[172.16.0.4:23:O]
<13:48:33.553> Thread.run:[172.16.0.4:23:O]
<13:48:33.658> Thread.run:[172.16.0.4:23:O][1649958513.646960][2620:2620] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
<13:48:33.659> Thread.run:[172.16.0.4:23:O][1649958513.647183][2620:2620] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_device_info.ini
<13:48:33.659> Thread.run:[172.16.0.4:23:O][1649958513.647513][2620:2620] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
<13:48:33.660> Thread.run:[172.16.0.4:23:O][1649958513.647658][2620:2620] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
<13:48:33.661> Thread.run:[172.16.0.4:23:O][1649958513.647758][2620:2620] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
<13:48:33.661> Thread.run:[172.16.0.4:23:O][1649958513.648034][2620:2620] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-R9Phbl)
<13:48:33.662> Thread.run:[172.16.0.4:23:O][1649958513.648657][2620:2620] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
<13:48:33.662> Thread.run:[172.16.0.4:23:O][1649958513.648703][2620:2620] CHIP:DL: NVS set: chip-counters/reboot-count = 2 (0x2)
<13:48:33.662> Thread.run:[172.16.0.4:23:O][1649958513.649209][2620:2620] CHIP:DL: Got Ethernet interface: eth0
<13:48:33.663> Thread.run:[172.16.0.4:23:O][1649958513.649518][2620:2620] CHIP:DL: Found the primary Ethernet interface:eth0
<13:48:33.663> Thread.run:[172.16.0.4:23:O][1649958513.649842][2620:2620] CHIP:DL: Got WiFi interface: wlan0
<13:48:33.663> Thread.run:[172.16.0.4:23:O][1649958513.651614][2620:2620] CHIP:DL: Found the primary WiFi interface:wlan0
<13:48:33.663> Thread.run:[172.16.0.4:23:O][1649958513.651719][2620:2620] CHIP:IN: UDP::Init bind&listen port=0
<13:48:33.663> Thread.run:[172.16.0.4:23:O][1649958513.651843][2620:2620] CHIP:IN: UDP::Init bound to port=55859
<13:48:33.663> Thread.run:[172.16.0.4:23:O][1649958513.651872][2620:2620] CHIP:IN: UDP::Init bind&listen port=0
<13:48:33.664> Thread.run:[172.16.0.4:23:O][1649958513.651969][2620:2620] CHIP:IN: UDP::Init bound to port=41133
<13:48:33.664> Thread.run:[172.16.0.4:23:O][1649958513.651994][2620:2620] CHIP:IN: BLEBase::Init - setting/overriding transport
<13:48:33.664> Thread.run:[172.16.0.4:23:O][1649958513.652017][2620:2620] CHIP:IN: TransportMgr initialized
<13:48:33.664> Thread.run:[172.16.0.4:23:O][1649958513.652054][2620:2620] CHIP:DIS: Init fabric pairing table with server storage
<13:48:33.664> Thread.run:[172.16.0.4:23:O][1649958513.652193][2620:2620] CHIP:IN: Loading from storage for fabric index 1
<13:48:33.665> Thread.run:[172.16.0.4:23:O][1649958513.654928][2620:2620] CHIP:IN: Loading from storage for fabric index 2
<13:48:33.665> Thread.run:[172.16.0.4:23:O][1649958513.655570][2620:2620] CHIP:IN: Loading from storage for fabric index 3
<13:48:33.665> Thread.run:[172.16.0.4:23:O][1649958513.656160][2620:2620] CHIP:DIS: Add fabric pairing table delegate
<13:48:33.665> Thread.run:[172.16.0.4:23:O][1649958513.656454][2620:2620] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-7Zxpqi)
<13:48:33.665> Thread.run:[172.16.0.4:23:O][1649958513.657053][2620:2620] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
<13:48:33.665> Thread.run:[172.16.0.4:23:O][1649958513.657105][2620:2620] CHIP:DL: NVS set: chip-counters/GlobalMCTR = 2000 (0x7D0)
<13:48:33.666> Thread.run:[172.16.0.4:23:O][1649958513.660495][2620:2620] CHIP:ZCL: Using ZAP configuration...
<13:48:33.666> Thread.run:[172.16.0.4:23:O][1649958513.663864][2620:2620] CHIP:DL: MDNS failed to join multicast group on wpan0 for address type IPv4: ../../examples/chip-tool/third_party/connectedhomeip/src/inet/UDPEndPointImplSockets.cpp:772: Inet Error 0x00000110: Address not found
<13:48:33.666> Thread.run:[172.16.0.4:23:O][1649958513.664177][2620:2620] CHIP:CTL: System State Initialized...
<13:48:33.666> Thread.run:[172.16.0.4:23:O][1649958513.664243][2620:2620] CHIP:CTL: Setting attestation nonce to random value
<13:48:33.666> Thread.run:[172.16.0.4:23:O][1649958513.664309][2620:2620] CHIP:CTL: Setting CSR nonce to random value
<13:48:33.667> Thread.run:[172.16.0.4:23:O][1649958513.664390][2620:2620] CHIP:IN: UDP::Init bind&listen port=5550
<13:48:33.667> Thread.run:[172.16.0.4:23:O][1649958513.664478][2620:2620] CHIP:IN: UDP::Init bound to port=5550
<13:48:33.667> Thread.run:[172.16.0.4:23:O][1649958513.664507][2620:2620] CHIP:IN: UDP::Init bind&listen port=5550
<13:48:33.667> Thread.run:[172.16.0.4:23:O][1649958513.664589][2620:2620] CHIP:IN: UDP::Init bound to port=5550
<13:48:33.667> Thread.run:[172.16.0.4:23:O][1649958513.664614][2620:2620] CHIP:IN: TransportMgr initialized
<13:48:33.668> Thread.run:[172.16.0.4:23:O][1649958513.664686][2620:2620] CHIP:CTL: Setting attestation nonce to random value
<13:48:33.668> Thread.run:[172.16.0.4:23:O][1649958513.664738][2620:2620] CHIP:CTL: Setting CSR nonce to random value
<13:48:33.668> Thread.run:[172.16.0.4:23:O][1649958513.665732][2620:2620] CHIP:CTL: Generating NOC
<13:48:33.668> Thread.run:[172.16.0.4:23:O][1649958513.666748][2620:2620] CHIP:DIS: Verifying the received credentials
<13:48:33.668> Thread.run:[172.16.0.4:23:O][1649958513.668711][2620:2620] CHIP:DIS: Added new fabric at index: 1, Initialized: 1
<13:48:33.669> Thread.run:[172.16.0.4:23:O][1649958513.668850][2620:2620] CHIP:DIS: Assigned compressed fabric ID: 0x9CBA16EBFF456F49, node ID: 0x000000000001B669
<13:48:33.669> Thread.run:[172.16.0.4:23:O][1649958513.668880][2620:2620] CHIP:CTL: Joined the fabric at index 1. Compressed fabric ID is: 0x9CBA16EBFF456F49
<13:48:33.669> Thread.run:[172.16.0.4:23:O][1649958513.668925][2620:2620] CHIP:IN: UDP::Init bind&listen port=5550
<13:48:33.669> Thread.run:[172.16.0.4:23:O][1649958513.669049][2620:2620] CHIP:IN: UDP::Init bound to port=5550
<13:48:33.669> Thread.run:[172.16.0.4:23:O][1649958513.669076][2620:2620] CHIP:IN: UDP::Init bind&listen port=5550
<13:48:33.670> Thread.run:[172.16.0.4:23:O][1649958513.669162][2620:2620] CHIP:IN: UDP::Init bound to port=5550
<13:48:33.670> Thread.run:[172.16.0.4:23:O][1649958513.669187][2620:2620] CHIP:IN: TransportMgr initialized
<13:48:33.670> Thread.run:[172.16.0.4:23:O][1649958513.669284][2620:2620] CHIP:CTL: Setting attestation nonce to random value
<13:48:33.670> Thread.run:[172.16.0.4:23:O][1649958513.669345][2620:2620] CHIP:CTL: Setting CSR nonce to random value
<13:48:33.670> Thread.run:[172.16.0.4:23:O][1649958513.670293][2620:2620] CHIP:CTL: Generating NOC
<13:48:33.671> Thread.run:[172.16.0.4:23:O][1649958513.671258][2620:2620] CHIP:DIS: Verifying the received credentials
<13:48:33.671> Thread.run:[172.16.0.4:23:O][1649958513.673047][2620:2620] CHIP:DIS: Added new fabric at index: 2, Initialized: 1
<13:48:33.671> Thread.run:[172.16.0.4:23:O][1649958513.673077][2620:2620] CHIP:DIS: Assigned compressed fabric ID: 0x17BBDE630F5825AD, node ID: 0x000000000001B669
<13:48:33.671> Thread.run:[172.16.0.4:23:O][1649958513.673101][2620:2620] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0x17BBDE630F5825AD
<13:48:33.672> Thread.run:[172.16.0.4:23:O][1649958513.673139][2620:2620] CHIP:IN: UDP::Init bind&listen port=5550
<13:48:33.672> Thread.run:[172.16.0.4:23:O][1649958513.673249][2620:2620] CHIP:IN: UDP::Init bound to port=5550
<13:48:33.672> Thread.run:[172.16.0.4:23:O][1649958513.673275][2620:2620] CHIP:IN: UDP::Init bind&listen port=5550
[... 204 lines cut ...]
<13:49:03.727> Thread.run:[172.16.0.4:23:O][1649958543.715263][2620:2620] CHIP:CTL: Shutting down the controller
<13:49:03.729> Thread.run:[172.16.0.4:23:O][1649958543.715392][2620:2620] CHIP:IN: Expiring all connections for fabric 1!!
<13:49:03.731> Thread.run:[172.16.0.4:23:O][1649958543.715481][2620:2620] CHIP:CTL: Shutting down the commissioner
[... cut ...]
<13:49:03.747> Thread.run:[172.16.0.4:23:O][1649958543.719161][2620:2620] CHIP:DL: Inet Layer shutdown
<13:49:03.748> Thread.run:[172.16.0.4:23:O][1649958543.719209][2620:2620] CHIP:DL: BLE shutdown
<13:49:03.749> Thread.run:[172.16.0.4:23:O][1649958543.719259][2620:2620] CHIP:DL: System Layer shutdown
<13:49:03.750> Thread.run:[172.16.0.4:23:O][1649958543.719471][2620:2620] CHIP:TOO: Run command failure: ../../examples/chip-tool/commands/common/CHIPCommand.cpp:391: CHIP Error 0x00000032: Timeout

 #### Proposed Solution
<suggested fix, suggested enhancement>

bzbarsky-apple, 2022-04-22:
@yinyihu-silabs Please attach the full log of the command that was used to commission the device.

yinyihu-silabs, 2022-04-25:
Here is the log for both commissioning and chip-tool read command when it get timeout error
[timeout commision.log](https://github.com/project-chip/connectedhomeip/files/8558013/timeout.commision.log)

yinyihu-silabs, 2022-04-25:
this timeout does not always happen right after commissioning.
at commit 077e44eadf46f349916b3f4f358da6126be2ed48, it happens after every 5-10 chip-tool commands (read/write attributes or functional commands)

bzbarsky-apple, 2022-04-26:
It looks like the Sigma1 message never got a response, even though it's the same IP as what was discovered at the end of commissioning and the Sigma handshake there worked fine.  Is it possible to attach both chip-tool and device logs from the same run (complete logs for both)?  Would be good to know what the device thinks it's doing.

yinyihu-silabs, 2022-04-26:
we noticed these two issues seen on all-cluster-app could be similar to what I'm seeing here on EFR32. https://github.com/project-chip/connectedhomeip/issues/17698
https://github.com/project-chip/connectedhomeip/issues/17568

bzbarsky-apple, 2022-04-27:
So the logs from https://github.com/project-chip/connectedhomeip/issues/17406#issuecomment-1110087952 show that client sent Sigma1, server received Sigma1 and responded with a standalone ack, but never sent Sigma2.  In particular, `CASEServer::InitCASEHandshake` failed on the server:
```
<info  > [IN] CASE Server received Sigma1 message. Starting handshake. EC 0x2001bcf0
<info  > [IN] CASE Server enabling CASE session setups
```
Compare to a working case, which looks more like this:
```
<info  > [IN] CASE Server received Sigma1 message. Starting handshake. EC 0x2001bcf0
<detail> [SC] Waiting for Sigma1 msg
<info  > [IN] CASE Server disabling CASE session setups
<info  > [SC] Received Sigma1 msg
...etc
```
we're not logging detailed error info on this device, looks like, but if this is in a build where #17698 could happen, that would have these symptoms.  And looking at the client log, it's the 16th read that fails, which _definitely_ starts to look like #17698.

yinyihu-silabs, 2022-06-08:
issue hasn't been seen for a while. close for now.