project-chip/connectedhomeip #17432: Error log message "MediaPlayback NOT returning ContentApp delegate for endpoint:1" observer in M5 stack logs

phonnakasturi-apple, 2022-04-15:
While testing the reporting feature on CHIPTool app with commit id 0680f217ac7313e5df56ae90946e45206515729e

MediaPlayback NOT returning ContentApp delegate for endpoint:1 error is getting logged.

Reporting works fine but the log displays the above mentioned error log messages.

Steps to reproduce:

1. git clone --recurse-submodules https://github.com/project-chip/connectedhomeip.git
2. cd connectedhomeip/
3. git reset --hard 0680f217ac7313e5df56ae90946e45206515729e
4. it submodule update --init
5. source scripts/bootstrap.sh
6. source scripts/activate.sh
7. cd ${HOME}/tools/esp-idf
8. ./install.sh
9. . ./export.sh
10. Flash M5 device
11. Install ChipTool App on iOS device
12. Launch the CHIPTool app
13. Provide the wifi configuration
14. Click on Temperature Sensor
15. Enter the values for Min Interval(sec) and Max Interval(sec)
16. Hit Send reporting settings

Attaching the complete log file for reference:
[Reporting error.txt](https://github.com/project-chip/connectedhomeip/files/8498847/Reporting.error.txt)

bzbarsky-apple, 2022-04-18:
Yes, the media stuff is not really hooked up in all-clusters-app...

woody-apple, 2022-06-29:
Agree, this should be tested using a media endpoint app

closed 2022-06-29 by woody-apple