project-chip/connectedhomeip #17456: ColorTemperatureMireds range is wrong

pankore, 2022-04-18:
#### Problem
* According to spec, the ColorTemperatureMireds should be ranging from 0 to 0xfeff
* Also from spec, actual color temperature = 1000000 / ColorTemperatureMireds
* However, running `./chip-tool colorcontrol move-to-color-temperature 256 0 0 0 1234 1` will make the color temperature loop over to 0
* It is changing the color temperature attribute instead of ColorTemperatureMireds. Related to #12372.

bzbarsky-apple, 2022-04-18:
I can't reproduce this problem....  Specifically, if I run the following commands after commissioning a node running chip-all-clusters-app with node id 1234:
```
chip-tool onoff on 1234 1
chip-tool colorcontrol read color-temperature 1234 1
chip-tool colorcontrol move-to-color-temperature 256 0 0 0 1234 1
chip-tool colorcontrol read color-temperature 1234 1
```
I see a color temperature of 0 after the first read and 256 after the second read.  @pankore Is it possible you were testing with the on/off bits set to off?

pankore, 2022-04-19:
Thank you. I will close this issue.

closed 2022-04-19 by pankore