project-chip/connectedhomeip #17460: [TC-MC-1.11] (Content Launcher cluster) attribute value mismatch

manjunathambiger23, 2022-04-18:
Summary:
As per testplan TC-MC-1.11 AcceptedCommandList and GeneratedCommandList attribute showing value mismatch error when executed
1. AcceptedCommandList expected value [0] but got [0,1].
2. GeneratedCommandList expected value [1] but got [ ] 0 entries.

Testplan screenshot: 
[image]

Execution screenshot and logs:
[MC11_ACL.txt](https://github.com/project-chip/connectedhomeip/files/8505642/MC11_ACL.txt)
[MC11_GCL.txt](https://github.com/project-chip/connectedhomeip/files/8505643/MC11_GCL.txt)
[image]
[image]

bzbarsky-apple, 2022-04-20:
This depends on the feature map of the cluster.  The generated command list must be `[2]`, but the accepted command list might be different things.  Please get the test plan fixed @manjunathambiger23

manjunathambiger23, 2022-04-21:
https://github.com/CHIP-Specifications/chip-test-plans/issues/1107

franck-apple, 2022-04-25:
> CHIP-Specifications/chip-test-plans#1107

test plan modified. Closing the issue.

closed 2022-04-25 by franck-apple