project-chip/connectedhomeip #17461: [TC-MC-1.12] (Account Login cluster) attribute value mismatch

manjunathambiger23, 2022-04-18:
Summary:
As per testplan TC-MC-1.12 AcceptedCommandList and GeneratedCommandList attribute got value mismatch error when executed
1.AcceptedCommandList expected value [0,2,3] but got [0
2.GeneratedCommandList expected value [1] but got [0]

Testplan screenshot:
[image]

Execution screenshot:
[image]
[MC12_ACL.txt](https://github.com/project-chip/connectedhomeip/files/8505683/MC12_ACL.txt)
[image]
[MC12_GCL.txt](https://github.com/project-chip/connectedhomeip/files/8505684/MC12_GCL.txt)

manjunathambiger23, 2022-04-21:
https://github.com/CHIP-Specifications/chip-test-plans/issues/1107

bzbarsky-apple, 2022-05-05:
#17833 might fix this.

manjunath-grl, 2022-05-10:
This issue was created by me using a account that i had to delete due to access issues, I will continue to follow these issues with my current id.

bzbarsky-apple, 2022-05-19:
At this point the accepted list is `[0]` and the generated list is `[1]`.  The actual ZAP config has everything disabled (not valid, for required commands), so I'm a little surprised the lists are not just empty...

woody-apple, 2022-05-27:
SDK Spec Review: Tagging @cjandhyala to follow up to see if this should be a cert blocker.

cjandhyala, 2022-05-30:
closing per @manjunath-grl request

closed 2022-05-30 by cjandhyala

manjunath-grl, 2022-06-03:
@cjandhyala its related to scripting issues can you please reopon it.

bzbarsky-apple, 2022-06-04:
@manjunath-grl I am happy to reopen if there is an updated description of the state of things on ToT right now and what's not working right.  Can you provide that please?  This issue is not actionable in the form it's in right now.