project-chip/connectedhomeip #17476: SessionManager test-only API improvement.

kghost, 2022-04-18:
#### Problem
Following up of issue #17411

* [ ] Use a macro define to guard test-only API: `SessionManager::InjectPaseSessionWithTestKey` and `SecureSessionTable::CreateNewSecureSessionForTest`
* [ ] Add test-only API to inject a CASE session into session manager.

kghost, 2022-04-21:
> Use a macro define to guard test-only API: `SessionManager::InjectPaseSessionWithTestKey` and `SecureSessionTable::CreateNewSecureSessionForTest`

This is probably not possible because tests shares the same lib with normal builds

@bzbarsky-apple

bzbarsky-apple, 2022-04-21:
> This is probably not possible because tests shares the same lib with normal builds

And yet `CONFIG_IM_BUILD_FOR_UNIT_TEST` works...  It's defined in the "standalone" build looks like, so maybe it depends on what one thinks a "normal build" is.