project-chip/connectedhomeip #17478: Timeout error is observed randomly while pairing thread accessory (Nordic board)

phonnakasturi-apple, 2022-04-19:
Using the SHA https://github.com/project-chip/connectedhomeip/commit/b4e425755d76efcaee1ebd0f82d5aac34c842474

While executing stress tests-> pairing with thread accessory, randomly pairing fails with Timeout error.
This issue was encountered twice.

**Steps to reproduce;**

1. Compile and flash thread accessory (nordic board)

2. Compile CHIPTool cli as follows
cd connectedhomeip
source scripts/bootstrap.sh
source scripts/activate.sh
gn gen out/debug --args='chip_mdns="platform" chip_crypto="mbedtls" chip_inet_config_enable_ipv4=false'
ninja -C out/debug
sudo ./out/debug/chip-tool pairing ble-thread node-id hex:operationalDataset setup-pin-code discriminator --pat-trust-store-path

**Following is the error;**

sudo ./out/debug/chip-tool pairing ble-thread 7 hex:0E080000000000000000000300001935060004001FFFC00208327870B273AE41620708FD327870B273000005104179F66B6F0CFDF7C2633F695CA48D91030E4D79486F6D653634363636303732010294C004102F56761D24FAD895F7A2D76C022906EC0C0402A0FFF8 20202021 3840 --paa-trust-store-path ./credentials/development/paa-root-certs
[1650321017844] [36590:4696752] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1650321017858] [36590:4696752] CHIP: [DL] _Init
[1650321017858] [36590:4696752] CHIP: [IN] UDP::Init bind&listen port=0
[1650321017858] [36590:4696752] CHIP: [IN] UDP::Init bound to port=52150
[1650321017858] [36590:4696752] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1650321017858] [36590:4696752] CHIP: [IN] TransportMgr initialized
[1650321017859] [36590:4696752] CHIP: [DIS] Init fabric pairing table with server storage
[1650321017859] [36590:4696752] CHIP: [IN] Loading from storage for fabric index 0x1
[1650321017860] [36590:4696752] CHIP: [IN] Loading from storage for fabric index 0x2
[1650321017860] [36590:4696752] CHIP: [IN] Loading from storage for fabric index 0x3
[1650321017860] [36590:4696752] CHIP: [DIS] Add fabric pairing table delegate
[1650321017862] [36590:4696752] CHIP: [ZCL] Using ZAP configuration...
[1650321017862] [36590:4696752] CHIP: [CTL] System State Initialized...
[1650321017863] [36590:4696752] CHIP: [CTL] Setting attestation nonce to random value
[1650321017863] [36590:4696752] CHIP: [CTL] Setting CSR nonce to random value
[1650321017863] [36590:4696752] CHIP: [CTL] Setting attestation nonce to random value
[1650321017863] [36590:4696752] CHIP: [CTL] Setting CSR nonce to random value
[1650321017866] [36590:4696752] CHIP: [CTL] Generating NOC
[1650321017868] [36590:4696752] CHIP: [DIS] Verifying the received credentials
[1650321017879] [36590:4696752] CHIP: [DIS] Added new fabric at index: 0x1, Initialized: 1
[1650321017879] [36590:4696752] CHIP: [DIS] Assigned compressed fabric ID: 0x732DA42E801C7C2D, node ID: 0x000000000001B669
[1650321017879] [36590:4696752] CHIP: [CTL] Joined the fabric at index 1. Compressed fabric ID is: 0x732DA42E801C7C2D
[1650321017879] [36590:4696752] CHIP: [CTL] Setting attestation nonce to random value
[1650321017879] [36590:4696752] CHIP: [CTL] Setting CSR nonce to random value
[1650321017882] [36590:4696752] CHIP: [CTL] Generating NOC
[1650321017884] [36590:4696752] CHIP: [DIS] Verifying the received credentials
[1650321017894] [36590:4696752] CHIP: [DIS] Added new fabric at index: 0x2, Initialized: 1
[1650321017894] [36590:4696752] CHIP: [DIS] Assigned compressed fabric ID: 0xE6E75F40B8EDD62F, node ID: 0x000000000001B669
[1650321017894] [36590:4696752] CHIP: [CTL] Joined the fabric at index 2. Compressed fabric ID is: 0xE6E75F40B8EDD62F
[1650321017894] [36590:4696752] CHIP: [CTL] Setting attestation nonce to random value
[1650321017894] [36590:4696752] CHIP: [CTL] Setting CSR nonce to random value
[1650321017897] [36590:4696752] CHIP: [CTL] Generating NOC
[1650321017899] [36590:4696752] CHIP: [DIS] Verifying the received credentials
[1650321017909] [36590:4696752] CHIP: [DIS] Added new fabric at index: 0x3, Initialized: 1
[1650321017909] [36590:4696752] CHIP: [DIS] Assigned compressed fabric ID: 0xD93825D0690DFEF3, node ID: 0x000000000001B669
[1650321017909] [36590:4696752] CHIP: [CTL] Joined the fabric at index 3. Compressed fabric ID is: 0xD93825D0690DFEF3
[1650321017922] [36590:4696755] CHIP: [DL] _OnPlatformEvent
[1650321017922] [36590:4696755] CHIP: [BLE] NewConnection
[1650321017922] [36590:4696755] CHIP: [SC] Assigned local session key ID 36765
[1650321017922] [36590:4696755] CHIP: [SC] Including MRP parameters in PBKDF param request
[1650321017922] [36590:4696755] CHIP: [IN] Prepared unauthenticated message 0x16f155588 to 0x0000000000000000 (0)  of type 0x20 and protocolId (0, 0) on exchange 56661i with MessageCounter:1692233578.
[1650321017922] [36590:4696755] CHIP: [IN] Sending unauthenticated msg 0x16f155588 with MessageCounter:1692233578 to 0x0000000000000000 at monotonic time: 000000001FF0A600 msec
[1650321017922] [36590:4696755] CHIP: [IN] Message appended to BLE send queue
[1650321017922] [36590:4696755] CHIP: [SC] Sent PBKDF param request
[1650321017922] [36590:4696755] CHIP: [CTL] Setting thread operational dataset from parameters
[1650321017922] [36590:4696755] CHIP: [CTL] Setting attestation nonce to random value
[1650321017922] [36590:4696755] CHIP: [CTL] Setting CSR nonce to random value
[1650321017922] [36590:4696755] CHIP: [CTL] Commission called for node ID 0x0000000000000007
[1650321017939] [36590:4696756] CHIP: [BLE] CBManagerState: ON
[1650321018315] [36590:4696755] CHIP: [BLE] Connecting to device with discriminator: 3840
[... 701 lines cut ...]
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the commissioner
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the controller
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the commissioner
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the controller
[1650321077011] [36590:4696752] CHIP: [IN] Expiring all connections for fabric 1!!
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the commissioner
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the controller
[1650321077011] [36590:4696752] CHIP: [IN] Expiring all connections for fabric 2!!
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the commissioner
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the controller
[1650321077011] [36590:4696752] CHIP: [IN] Expiring all connections for fabric 3!!
[1650321077011] [36590:4696752] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1650321077011] [36590:4696752] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321077011] [36590:4696752] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321077011] [36590:4696752] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321077011] [36590:4696752] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321077011] [36590:4696752] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1650321077012] [36590:4696752] CHIP: [BLE] CancelConnection
[1650321077012] [36590:4696752] CHIP: [IN] Clearing BLE pending packets.
[1650321077012] [36590:4696752] CHIP: [BLE] BtpEngine unsub failed
[1650321077012] [36590:4696752] CHIP: [BLE] Auto-closing end point's BLE connection.
[1650321077012] [36590:4696752] CHIP: [DL] Inet Layer shutdown
[1650321077012] [36590:4696752] CHIP: [DL] BLE shutdown
[1650321077012] [36590:4696752] CHIP: [DL] System Layer shutdown
[1650321077012] [36590:4696752] CHIP: [TOO] Run command failure: ../../src/protocols/secure_channel/CASESession.cpp:266: CHIP Error 0x00000032: Timeout

bzbarsky-apple, 2022-04-19:
Looks like the server never responded to the Sigma1 message....  @phonnakasturi-apple Did we resolve the right IP for this server, do you know?

phonnakasturi-apple, 2022-04-21:
They were on the right network. If there is a specific command that needs to be executed then please ket me know, I can verify

bzbarsky-apple, 2022-04-21:
@phonnakasturi-apple Well, the resolved IP was fdb0:a47c:fc46:0:9d1:9a28:c383:3c7 and it was on interface en7.

So checking whether `ping6 fdb0:a47c:fc46:0:9d1:9a28:c383:3c7` and `ping6 fdb0:a47c:fc46:0:9d1:9a28:c383:3c7%en7` work would be a good start.  Assuming the thread accessory still has the same IP...

If it does not have the same IP, and if this is somewhat reproducible, the next time this happens please look for the log line that looks like:
```
 [1650321026998] [36590:4696755] CHIP: [SC] Establishing CASE session to UDP:[fdb0:a47c:fc46:0:9d1:9a28:c383:3c7]:5540
```
and see whether that IP is ping-able from the device running chip-tool.

phonnakasturi-apple, 2022-06-22:
@bzbarsky-apple I saw this issue again with SHA 756e3abe620e634095bac7609b1178f677f7cbf7

And I try to ping the ip I see no output;

% ping6 fdb0:a47c:fc46:0:6b96:a2f0:d434:1667   
PING6(56=40+8+8 bytes) fd78:70b2:73ae:4162:ccf:99:5bef:d6fa --> fdb0:a47c:fc46:0:6b96:a2f0:d434:1667

bzbarsky-apple, 2022-06-22:
@phonnakasturi-apple does `ping6 fdb0:a47c:fc46:0:6b96:a2f0:d434:1667%en7` or whatever the right interface is work?

It sounds like there are some sort of networking issues here that should get sorted out in the specific setup you are using.

phonnakasturi-apple, 2022-08-08:
@bzbarsky-apple  
Attaching the logs from chip-tool cli 
[nordicLogs.txt](https://github.com/project-chip/connectedhomeip/files/9285595/nordicLogs.txt)

and the ping command is 
ping6 fd6d:280f:454d:0:e70f:dee0:c20f:28d8%en10

bzbarsky-apple, 2022-08-09:
Per offline discussion: something weird is going on with ping6 for @phonnakasturi-apple in general.  Instead, we'll try to capture logs from both sides in the failing case and see whether the two sides are dealing with the same IP.

closed 2023-02-19 by stale[bot]