project-chip/connectedhomeip #17479: Segmentation failure occurs after Timeout issue (#17478) while pairing thread accessory (Nordic board)

phonnakasturi-apple, 2022-04-19:
Using the SHA https://github.com/project-chip/connectedhomeip/commit/b4e425755d76efcaee1ebd0f82d5aac34c842474

While executing stress tests-> pairing with thread accessory, segmentation error is thrown soon after Time out issue #17478
This issue was observed twice and both the times it followed Timeout error mentioned in the issue #17478

**Steps to reproduce;**

1. Compile and flash thread accessory (nordic board)

2. Compile CHIPTool cli as follows
cd connectedhomeip
source scripts/bootstrap.sh
source scripts/activate.sh
gn gen out/debug --args='chip_mdns="platform" chip_crypto="mbedtls" chip_inet_config_enable_ipv4=false'
ninja -C out/debug
sudo ./out/debug/chip-tool pairing ble-thread node-id hex:operationalDataset setup-pin-code discriminator --pat-trust-store-path

**Following is the error snippet ;**

[1650321457476] [36665:4701009] CHIP: [IN] Sending unauthenticated msg 0x16f3c51f8 with MessageCounter:3829564977 to 0x0000000000000000 at monotonic time: 000000001FF75B01 msec
[1650321457476] [36665:4701009] CHIP: [EM] Flushed pending ack for MessageCounter:2477388310 on exchange 14952i
[1650321457654] [36665:4701009] CHIP: [EM] Received message of type 0x10 with protocolId (0, 0) and MessageCounter:2477388311 on exchange 14952i
[1650321457655] [36665:4701009] CHIP: [EM] OnMessageReceived failed, err = ../../src/messaging/ExchangeMgr.cpp:269: CHIP Error 0x00000070: Unsolicited msg with originator bit clear
[1650321457783] [36665:4701010] CHIP: [EM] Received message of type 0x9 with protocolId (0, 1) and MessageCounter:13695313 on exchange 14953i
[1650321457783] [36665:4701010] CHIP: [EM] Found matching exchange: 14953i, Delegate: 0x12f804690
[1650321457783] [36665:4701010] CHIP: [EM] Rxd Ack; Removing MessageCounter:12138406 from Retrans Table on exchange 14953i
[1650321457783] [36665:4701010] CHIP: [EM] Removed CHIP MessageCounter:12138406 from RetransTable on exchange 14953i
[1650321457783] [36665:4701010] CHIP: [DMG] ICR moving to [ResponseRe]
[1650321457783] [36665:4701010] CHIP: [DMG] InvokeResponseMessage =
[1650321457783] [36665:4701010] CHIP: [DMG] {
[1650321457783] [36665:4701010] CHIP: [DMG] 	suppressResponse = false, 
[1650321457783] [36665:4701010] CHIP: [DMG] 	InvokeResponseIBs =
[1650321457783] [36665:4701010] CHIP: [DMG] 	[
[1650321457783] [36665:4701010] CHIP: [DMG] 		InvokeResponseIB =
[1650321457783] [36665:4701010] CHIP: [DMG] 		{
[1650321457783] [36665:4701010] CHIP: [DMG] 			CommandDataIB =
[1650321457783] [36665:4701010] CHIP: [DMG] 			{
[1650321457783] [36665:4701010] CHIP: [DMG] 				CommandPathIB =
[1650321457783] [36665:4701010] CHIP: [DMG] 				{
[1650321457783] [36665:4701010] CHIP: [DMG] 					EndpointId = 0x0,
[1650321457783] [36665:4701010] CHIP: [DMG] 					ClusterId = 0x30,
[1650321457783] [36665:4701010] CHIP: [DMG] 					CommandId = 0x5,
[1650321457783] [36665:4701010] CHIP: [DMG] 				},
[1650321457783] [36665:4701010] CHIP: [DMG] 				
[1650321457783] [36665:4701010] CHIP: [DMG] 				CommandData = 
[1650321457783] [36665:4701010] CHIP: [DMG] 				{
[1650321457783] [36665:4701010] CHIP: [DMG] 					0x0 = 0, 
[1650321457783] [36665:4701010] CHIP: [DMG] 					0x1 = "", 
[1650321457783] [36665:4701010] CHIP: [DMG] 				},
[1650321457783] [36665:4701010] CHIP: [DMG] 			},
[1650321457783] [36665:4701010] CHIP: [DMG] 			
[1650321457783] [36665:4701010] CHIP: [DMG] 		},
[1650321457783] [36665:4701010] CHIP: [DMG] 		
[1650321457783] [36665:4701010] CHIP: [DMG] 	],
[1650321457783] [36665:4701010] CHIP: [DMG] 	
[1650321457783] [36665:4701010] CHIP: [DMG] 	InteractionModelRevision = 1
[1650321457783] [36665:4701010] CHIP: [DMG] },
[1650321457783] [36665:4701010] CHIP: [DMG] Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0005
[1650321457783] [36665:4701010] CHIP: [CTL] Received CommissioningComplete response
[1650321457783] [36665:4701010] CHIP: [CTL] Successfully finished commissioning step 'SendComplete'
[1650321457783] [36665:4701010] CHIP: [CTL] Commissioning stage next step: 'SendComplete' -> 'Cleanup'
[1650321457783] [36665:4701010] CHIP: [CTL] Performing next commissioning step 'Cleanup'
[1650321457783] [36665:4701010] CHIP: [CTL] Successfully finished commissioning step 'Cleanup'
[1650321457783] [36665:4701010] CHIP: [TOO] Device commissioning completed with success
[1650321457783] [36665:4701010] CHIP: [DMG] ICR moving to [AwaitingDe]
[1650321457783] [36665:4701010] CHIP: [EM] Sending Standalone Ack for MessageCounter:13695313 on exchange 14953i
[1650321457783] [36665:4701010] CHIP: [IN] Prepared secure message 0x16f451238 to 0x0000000000000009 (1)  of type 0x10 and protocolId (0, 0) on exchange 14953i with MessageCounter:12138407.
[1650321457783] [36665:4701010] CHIP: [IN] Sending encrypted msg 0x16f451238 with MessageCounter:12138407 to 0x0000000000000009 (1) at monotonic time: 000000001FF75C35 msec
[1650321457783] [36665:4701010] CHIP: [EM] Flushed pending ack for MessageCounter:13695313 on exchange 14953i
[1650321457783] [36665:4701005] CHIP: [CTL] Shutting down the commissioner
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the controller
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the commissioner
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the controller
[1650321457784] [36665:4701005] CHIP: [IN] Expiring all connections for fabric 1!!
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the commissioner
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the controller
[1650321457784] [36665:4701005] CHIP: [IN] Expiring all connections for fabric 2!!
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the commissioner
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the controller
[1650321457784] [36665:4701005] CHIP: [IN] Expiring all connections for fabric 3!!
[1650321457784] [36665:4701005] CHIP: [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1650321457784] [36665:4701005] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321457784] [36665:4701005] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321457784] [36665:4701005] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321457784] [36665:4701005] CHIP: [DMG] IM WH moving to [Uninitialized]
[1650321457784] [36665:4701005] CHIP: [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1650321457784] [36665:4701005] CHIP: [BLE] CancelConnection
[1650321457784] [36665:4701005] CHIP: [IN] Clearing BLE pending packets.
[1650321457784] [36665:4701005] CHIP: [BLE] BtpEngine unsub failed
[1650321457784] [36665:4701005] CHIP: [BLE] Auto-closing end point's BLE connection.
zsh: segmentation fault  sudo ./out/debug/chip-tool pairing ble-thread 9  20202021 3840  

**Log Files:** 

Attaching the complete log files from CHIPTool cli and from nordic board:
[SegmentationError-CHIPToolcliLogs.txt](https://github.com/project-chip/connectedhomeip/files/8509145/SegmentationError-CHIPToolcliLogs.txt)

[SegmentationError-NordicBoardLogs.txt](https://github.com/project-chip/connectedhomeip/files/8509146/SegmentationError-NordicBoardLogs.txt)

bzbarsky-apple, 2022-04-19:
@phonnakasturi-apple Is it possible to run under lldb and get a stack trace?

phonnakasturi-apple, 2022-04-19:
@bzbarsky-apple Can you please let me know how to run thread accessory under lldb?

bzbarsky-apple, 2022-04-20:
The segmentation fault is in chip-tool, not the thread accessory.  So:
```
lldb ./out/debug/chip-tool -- pairing ble-thread node-id hex:operationalDataset setup-pin-code discriminator --pat-trust-store-path
```
and then when it crashes:
```
bt
```
to get the stack and:
```
f 0
```
to get some info about the topmost frame.

closed 2022-10-28 by stale[bot]