project-chip/connectedhomeip #17483: FeatureMap Attribute is not present in following clusters

kowsisoundhar12, 2022-04-19:
Feature Map global attribute is not present in following Clusters:

1.Identify Cluster
2.lluminance Measurement Cluster
3.Color Control Cluster
4.Flow Measurement Cluster
5.Occupancy Sensing Cluster
6.Pressure Measurement Cluster
[I-1-1-attribute-missing.zip](https://github.com/project-chip/connectedhomeip/files/8509699/I-1-1-attribute-missing.zip)
[ILL-1-1-Attribute-missing.zip](https://github.com/project-chip/connectedhomeip/files/8509700/ILL-1-1-Attribute-missing.zip)
[CC-1-1-ATtribute-missing.zip](https://github.com/project-chip/connectedhomeip/files/8509701/CC-1-1-ATtribute-missing.zip)
[FLW-1-1-Attribute-missing.zip](https://github.com/project-chip/connectedhomeip/files/8509702/FLW-1-1-Attribute-missing.zip)
[OCC-1-1-Attribute-missing.zip](https://github.com/project-chip/connectedhomeip/files/8509703/OCC-1-1-Attribute-missing.zip)
[PRS-1-1-Featuremap-missing.zip](https://github.com/project-chip/connectedhomeip/files/8509704/PRS-1-1-Featuremap-missing.zip)

manjunath-grl, 2022-05-02:
Feature Map global attribute is not present in following Clusters:
1.Power Source Configuration

[PSCFG11_FM.txt](https://github.com/project-chip/connectedhomeip/files/8605563/PSCFG11_FM.txt)

bzbarsky-apple, 2022-05-09:
Duplicate of #16725

closed 2022-05-09 by bzbarsky-apple