project-chip/connectedhomeip #17484: AcceptedCommandList, GeneratedCommandList Attributes are not present in following clusters

kowsisoundhar12, 2022-04-19:
AcceptedCommandList, GeneratedCommandList Attributes are missing in following cluster:
1.Pressure Measurement Cluster
2.Thermostat Cluster
[image]
[image]
[PRS-1-1-Missing-Attribute.zip](https://github.com/project-chip/connectedhomeip/files/8509785/PRS-1-1-Missing-Attribute.zip)
[TSTAT-1-1-Missing-Attribute.zip](https://github.com/project-chip/connectedhomeip/files/8509786/TSTAT-1-1-Missing-Attribute.zip)

bzbarsky-apple, 2022-04-22:
@kowsisoundhar12 What are the steps to reproduce this?  Running `zap_regen_all.py` on tip of Matter obviously does not run into problem (this runs in CI all the time), so what does one need to do to run into it?

raju-apple, 2022-04-26:
@kowsisoundhar12 Can you please attach the yaml to the issue as well to help reproduce it..