project-chip/connectedhomeip #17500: [TC-IDM-2.2]While verifying Read Request Message to the DUT from TH to read any attribute to an unsupported node throws an Run command failure instead UNSUPPORTED_NODE(0x9B)

Rajashreekalmane, 2022-04-19:
#### Problem
As per Test-Plan test step 13  While verifying Read Request Message to the DUT from TH to read any attribute to an unsupported node throws an Run command failure instead UNSUPPORTED_NODE(0x9B).

 * expected behavior:-UNSUPPORTED_NODE(0x9B)
 * actual behavior:-Run command failure

 * steps to reproduce:- 
While pairing provided nodeID as 1 
./chip-tool pairing ble-wifi 1 zigbee1 matter123 20202021 3840
To verify this step provided nodeID as 2
./chip-tool basic read product-name 2 0

 * system configuration
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: https://github.com/project-chip/connectedhomeip/commit/2982c0ca0623773327def4cc7180d07f6b959c42
Dut: All-Clusters-app

Test-Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-2-2-report-data-action-from-dut-to-th

Spec Reference:-
https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/data_model/Interaction-Model.adoc#ref_IM.StatusCodeTable

PFA log below:-
[Unsupported NodeID.txt](https://github.com/project-chip/connectedhomeip/files/8512792/Unsupported.NodeID.txt)

yunhanw-google, 2022-04-25:
Per jerry, https://github.com/project-chip/connectedhomeip/issues/16705#issuecomment-1104440723, so we need to close it.

yunhanw-google, 2022-04-25:
@bzbarsky-apple  since we are not supporting unsupported_node, I think timeout error is valid here?

bzbarsky-apple, 2022-04-27:
The test plan described above can't be tested with the 1.0 SDK at all.  

The test steps above, in particular, would _not_ lead to UNSUPPORTED_NODE even if that were implemented.  They would just lead to a discovery error, or no error at all, just talking to the wrong node.

A test step that _could_ lead to UNSUPPORTED_NODE would involve sending an actual Node id in a path that does not match the right Node id.  And as I said, the current SDK cannot do it.  And yes, the only place where this would be useful is proxies, and proxies are out of scope for 1.0, so it's not getting implemented for 1.0.

So that's the summary:

1. The test plan cannot be tested with the SDK.
2. The steps above do not test what the test plan wants to test.

Rajashreekalmane, 2022-04-27:
@bzbarsky-apple Thank you.

closed 2022-04-27 by Rajashreekalmane