project-chip/connectedhomeip #17510: Keypad Input cluster showing failure response for Endpoint 1

manjunathambiger23, 2022-04-19:
Summary:
Keypad Input cluster:  showing :" 0xc3 (UNSUPPORTED_CLUSTER) " response for endpoint "1" when executed.

Testplan screenshot:
[image]

bzbarsky-apple, 2022-04-20:
@manjunathambiger23 Is this being tested against an app (like tv-app) that actually supports this cluster?

manjunathambiger23, 2022-04-21:
@bzbarsky-apple yes its been tested against tv-app

manjunathambiger23, 2022-04-21:
we use the following command for Build
./scripts/run_in_build_env.sh \
                     "./scripts/build/build_examples.py \
                        --target linux-x64-chip-tool-no-ble-tsan \
                        --target linux-x64-all-clusters-test-group-no-ble-tsan \
                        --target linux-x64-lock-app-no-ble-tsan \
                        --target linux-x64-tv-app-no-ble-tsan \
                        build \
                        --copy-artifacts-to objdir-clone \
                     "

chrisdecenzo, 2022-04-22:
For some reason, it looks like the KeypadInput cluster commands are not enabled in zap:

        {
          "name": "Keypad Input",
          "code": 1289,
          "mfgCode": null,
          "define": "KEYPAD_INPUT_CLUSTER",
          "side": "server",
          "enabled": 1,
          "commands": [],
          "attributes": [
            {
              "name": "ClusterRevision",
              "code": 65533,
              "mfgCode": null,
              "side": "server",
              "included": 1,
              "storageOption": "RAM",
              "singleton": 0,
              "bounded": 0,
              "defaultValue": "1",
              "reportable": 1,
              "minInterval": 0,
              "maxInterval": 65344,
              "reportableChange": 0
            }
          ]
        },

When I perform a read of the cluster revision attribute on this cluster, I get back success:

./out/debug/chip-tool keypadinput read cluster-revision 1234 1

So I think the issue is just that the commands are not enabled.

chrisdecenzo, 2022-04-22:
And it looks like this never worked on all-clusters-app

chrisdecenzo, 2022-04-22:
hi @lazarkov - can you enable all the commands and all the attributes for each media cluster on the all-clusters-app please?

lazarkov, 2022-04-22:
@manjunathambiger23 can you please use tv-app for testing? all-clusters-app is not supported.

closed 2022-04-22 by lazarkov