project-chip/connectedhomeip #17530: [SQA] Thread  NetworkingStatus value shows NetworkIDNotFound.

Survensa, 2022-04-20:
**Problem:**
In (TC-DM-4.10) the NetworkingStatus shows the value 3(NetworkIDNotFound) for the same network ID of the thread network.

**Expected behavior:**  networkingStatus: 0

**Actual behavior:**  networkingStatus: 3

**Steps to reproduce:** `./chip-tool networkcommissioning  remove-network 0119100899082022 62 0`

**Platform:** Thread

**CommitID** used : 20e1a4a705770d5222a8ebf04e12ab2ce0304df6

**Referance** :  [spec](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/NetworkCommissioningCluster.adoc),[ Test-plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc) .

**Logs:**

> [1650391420.469341][10048:10053] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1650391420.469452][10048:10053] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1650391420.469634][10048:10053] CHIP:TOO:   NetworkConfigResponse: {
[1650391420.469746][10048:10053] CHIP:TOO:     networkingStatus: 3
[1650391420.469812][10048:10053] CHIP:TOO:    }
[1650391420.469902][10048:10053] CHIP:DMG: ICR moving to [AwaitingDe]

Damian-Nordic, 2022-04-25:
Try passing the network ID with `hex:` prefix. It works for me that way.

andy31415, 2022-04-25:
@Survensa - please validate. It sounds as if the network ID you have is `hex` even though by chance it happens to only contain digits. Can you confirm if this is correct and it passes?

Survensa, 2022-05-06:
Executed with `hex:` prefix getting the expected outcome.

closed 2022-05-06 by Survensa