project-chip/connectedhomeip #17544: [SQA] TC-DM-4.13 Reorder Network command - can't be validated as Networks attribute holds only 1 entry.

Survensa, 2022-04-20:
#### Problem
The Networks attribute has 1 entry and user tries to add more entries with AddOrUpdateWiFi command, the DUT responds with NetworkingStatus as '2' (BoundsExceeded) in NetworkResponse. Due to which the Networks attribute holds only 1 entry. 

In this case, Reorder Network command is send by setting NetworkIndex value as Eg:3. As Networks attribute has only 1 entry it's not possiable to set the NetworkID to the NetworkIndex 3. but still the DUT responded with NetworkIndex '3' and NetworkingStatus '0' (Success) in NetworkResponse.

### Questions:
1. Why we can't add  more networks to the network attribute list using AddOrUpdateWiFi command, when only 1 entry is present already? If there are any alternate ways to add networks, kindly let us know. 
2. When networks attribute hold only 1 entry, the Reorder Network command holds the changing NetworkIndex as 3. still why the DUT is responding with NetworkingStatus as Success?
 
### Actual behavior:
 * **steps to reproduce**

1. cmd `./chip-tool networkcommissioning add-or-update-wi-fi-network hex:47524C50726976617465 grlprivatewifi092010 1 0` to add **WiFinet_idx1**
2. cmd `./chip-tool networkcommissioning add-or-update-wi-fi-network hex:47524C507269766174655F455854 matter123 1 0` to add **WiFinet_idx2**
3. cmd `./chip-tool networkcommissioning add-or-update-wi-fi-network hex:7369766172616D2773206950686F6E65 maskmask 1 0` to add **WiFinet_idx3**
4. cmd `./chip-tool networkcommissioning reorder-network hex:47524C50726976617465 3 1 0` to reorder the Network.
5. cmd` ./chip-tool networkcommissioning read networks 1 0` to list add three network added.

 * **system configuration** - Used the commit 20e1a4a705770d5222a8ebf04e12ab2ce0304df6 on RPI with Wifi network
 *  Reference : https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/devicemanagement.adoc#test-procedure-23
 >

 #### Output log.
```
**./chip-tool networkcommissioning add-or-update-wi-fi-network hex:47524C50726976617465 grlprivatewifi092010 1 0**
[1650381475.218029][44068:44073] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1650381475.218092][44068:44073] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1650381475.218190][44068:44073] CHIP:TOO:   NetworkConfigResponse: {
[1650381475.218252][44068:44073] CHIP:TOO:     networkingStatus: 0
[1650381475.218289][44068:44073] CHIP:TOO:     networkIndex: 0
[1650381475.218323][44068:44073] CHIP:TOO:    }
[1650381475.218376][44068:44073] CHIP:DMG: ICR moving to [AwaitingDe]
[1650381475.218447][44068:44073] CHIP:EM: Sending Standalone Ack for MessageCounter:7658401 on exchange 620i

**/chip-tool networkcommissioning add-or-update-wi-fi-network hex:47524C507269766174655F455854 matter123 1 0**

[1650381481.251038][44074:44079] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1650381481.251102][44074:44079] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1650381481.251189][44074:44079] CHIP:TOO:   NetworkConfigResponse: {
[1650381481.251250][44074:44079] CHIP:TOO:     networkingStatus: 2
[1650381481.251289][44074:44079] CHIP:TOO:    }
[1650381481.251342][44074:44079] CHIP:DMG: ICR moving to [AwaitingDe]
[1650381481.251412][44074:44079] CHIP:EM: Sending Standalone Ack for MessageCounter:11373968 on exchange 55916i

**./chip-tool networkcommissioning add-or-update-wi-fi-network hex:7369766172616D2773206950686F6E65 maskmask 1 0**
[1650381487.889851][44081:44086] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1650381487.889909][44081:44086] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1650381487.889987][44081:44086] CHIP:TOO:   NetworkConfigResponse: {
[1650381487.890043][44081:44086] CHIP:TOO:     networkingStatus: 2
[1650381487.890079][44081:44086] CHIP:TOO:    }
[1650381487.890127][44081:44086] CHIP:DMG: ICR moving to [AwaitingDe]
[1650381487.890188][44081:44086] CHIP:EM: Sending Standalone Ack for MessageCounter:3375326 on exchange 62297i

 **./chip-tool networkcommissioning reorder-network hex:47524C50726976617465 3 1 0**
[1650381494.820649][44087:44092] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0031 Command=0x0000_0005
[1650381494.820749][44087:44092] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Command 0x0000_0005
[1650381494.820860][44087:44092] CHIP:TOO:   NetworkConfigResponse: {
[1650381494.820940][44087:44092] CHIP:TOO:     networkingStatus: 0
[1650381494.820979][44087:44092] CHIP:TOO:     networkIndex: 3
[1650381494.821016][44087:44092] CHIP:TOO:    }
[1650381494.821087][44087:44092] CHIP:DMG: ICR moving to [AwaitingDe]
[1650381494.821174][44087:44092] CHIP:EM: Sending Standalone Ack for MessageCounter:5656346 on exchange 4849i

**./chip-tool networkcommissioning read networks 1 0**
[1650381501.447787][44093:44098] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0031 Attribute 0x0000_0001 DataVersion: 3182776035
[1650381501.447899][44093:44098] CHIP:TOO:   Networks: 1 entries
[1650381501.447965][44093:44098] CHIP:TOO:     [1]: {
[1650381501.448003][44093:44098] CHIP:TOO:       NetworkID: 47524C50726976617465
[1650381501.448037][44093:44098] CHIP:TOO:       Connected: FALSE
[1650381501.448071][44093:44098] CHIP:TOO:      }
[1650381501.448221][44093:44098] CHIP:EM: Sending Standalone Ack for MessageCounter:10492805 on exchange 43057i
```

andy31415, 2022-04-25:
@Survensa trying to figure out if this is a mandatory requirement or not. At a first glance, supporting a single network (e.g. a single wifi network) sounds workable. Are there spec entries that mandate multiple network support? Could you reference the spec parts that describe minimum requirements here?

erjiaqing, 2022-04-27:
This is a intended behavior to simplify the platform driver for now. From the Spec, the minimal value of MaxNetworks can be 1, so it is spec compliance if support only one network.

closed 2022-05-10 by Survensa