project-chip/connectedhomeip #17569: Update existing SessionHolders to shift to a newly established session atomically

mrjerryjohns, 2022-04-20:
#### Problem

When a new session has been established to a peer and there are existing sessions to the same peer, we should 'atomically' shift any existing `SessionHolder`s over to the new session. This follows Tenet #3 in #17298.

This requires updating the existing `SessionDelegate` to add a new `OnSessionUpdated` API to swap out an old SessionHandle with a new one.

kghost, 2022-04-20:
For a `SessionHolders` able to smoothly shift to the new session, It requires that the `SessionDelegate` explicitly support such shifting, for example, `ExchangeContext` is not eligible for such shift, becasue exchanges are bound to a session.

For an delegate declare supporting the shift, it must use `SessionHolderWithDelegate` class, and override `OnSessionUpdated` callback, let it return `CHIP_NO_ERROR`. The default `OnSessionUpdated` will return a `CHIP_ERROR_NOT_SUPPORTED` error.

mrjerryjohns, 2022-04-20:
> for example, ExchangeContext is not eligible for such shift

Those that do not permit a shift can choose to do what they want with the new session. For ExchangeContext, it should continue to hold on to the existing session since it cannot actually move to the new session by spec rules. An `OperationalDeviceProxy` however should be able to move over.

>  let it return CHIP_NO_ERROR

Do we even need to define an error return for this new `OnSessionUpdated` callback? I'd think it can just be a void return, since the caller in this case doesn't get affected by what the callee's do with the new session.

kghost, 2022-04-20:
So delegate need to grab the new session inside `OnSessionUpdated` ? For example, it must implement this to support automatic shifting:

```
void SomeThing::OnSessionUpdated(const SessionHandle newSession) override
{
    mSessionOfSomeThing.Grab(newSession);
}
```

mrjerryjohns, 2022-04-20:
Yup, that looks good.

PR #18107 (2022-06-22): Implement SessionHolder auto shifting
#### Problem
Fix #17569

#### Change overview
Implement `SessionManager::ShiftToSession(const SessionHandle & handle)`
Shift old session the the same peer to a newly created session.

#### Testing
Passed unit-tests. Added new test case for `SessionManager::ShiftToSession`
closed 2022-06-22 by kghost