project-chip/connectedhomeip #17585: Support auto generation of optional arguments for yaml tests

carol-apple, 2022-04-21:
#### Problem
For tests ran in the test suites, cluster command arguments are passed in via the yaml file. There is a way to specify that an argument is optional by doing adding `isOptional : true`. Using SystemCommands as an example:

```
const Start = {
  name : 'Start',
  arguments : [
    { 'name' : 'discriminator', type : 'INT16U', isOptional : true }, { 'name' : 'port', type : 'INT16U', isOptional : true }, { 'name' : 'kvs', type : 'CHAR_STRING', isOptional : true }
  ],
  response : { arguments : [] }
};
```

However, this translates to the generated Commands.h to call a particular cluster command function by not supplying the argument, rather than making it truly optional. For instance, if discriminator is not supplied in the yaml file, the generated test code looks like:

```
CHIP_ERROR TestStartTheAccessoryWithNoCommandLineOptions_2()
    {
        SetIdentity(kIdentityAlpha);
        return Start();
    }
```

This works because SystemCommands.h declares the function like so:

```
CHIP_ERROR Start(uint16_t discriminator = 0xFFFF, uint16_t port = CHIP_PORT, const char * kvs = nullptr);
```

This is insufficient as the yaml file cannot just supply port or kvs, for example. It must supply all "optional" parameters that come before it.

 #### Proposed Solution
Make the generated test code in Commands.h to support signatures like:
```
CHIP_ERROR Start(Optional<uint16_t> discriminator, Optional<uint16_t> port, Optional<chip::CharSpan> kvs);
```

bzbarsky-apple, 2022-04-21:
@vivien-apple

bzbarsky-apple, 2022-05-22:
@carol-apple Is this still relevant given the other changes to these commands?

bzbarsky-apple, 2022-05-23:
Per offline conversation with @carol-apple: not relevant anymore.

closed 2022-05-23 by bzbarsky-apple

carol-apple, 2022-05-23:
For posterity, this PR added generation of simulated-cluster-objects.h which allows for arguments to be truly optional: https://github.com/project-chip/connectedhomeip/pull/18038