project-chip/connectedhomeip #17592: [SQA] THREAD-In (TC-PS-2.1) SDK needs implementation on non-global Mandatory attributes and optional attributes.

Survensa, 2022-04-21:
**problem:**
While Executing the  (TC-PS-2.1) non-global attributes of the power source cluster, few Mandatory attributes and optional attributes throws error code as 0x86(unsupported attribute). 

**MANDATORY:**
```

./chip-tool powersource read wired-current-type 1 1
./chip-tool powersource read battery-quantity 1 1
./chip-tool powersource read battery-functional-while-charging 1 1
./chip-tool powersource read battery-replacement-description 1 1
./chip-tool powersource read battery-replaceability 1 1
./chip-tool powersource read battery-replacement-needed 1 1
```

**OPTIONAL:**

```
./chip-tool powersource read wired-assessed-input-voltage 1 1
./chip-tool powersource read wired-assessed-input-frequency 1 1
./chip-tool powersource read wired-assessed-current 1 1
./chip-tool powersource read wired-nominal-voltage 1 1
./chip-tool powersource read wired-maximum-current 1 1
./chip-tool powersource read wired-present 1 1
./chip-tool powersource read active-wired-faults 1 1
./chip-tool powersource read battery-present 1 1
./chip-tool powersource read battery-common-designation 1 1
./chip-tool powersource read battery-ansidesignation 1 1
./chip-tool powersource read battery-iecdesignation 1 1
./chip-tool powersource read battery-approved-chemistry 1 1
./chip-tool powersource read battery-capacity 1 1
./chip-tool powersource read battery-time-to-full-charge 1 1
./chip-tool powersource read battery-charging-current 1 1
./chip-tool powersource read active-battery-charge-faults 1 1
```

**Expected behavior:** status 0x0

**Actual behavior:**  status = 0x86 (UNSUPPORTED_ATTRIBUTE)

**Platform:** Thread

**CommitID** used : 20e1a4a705770d5222a8ebf04e12ab2ce0304df6

**Reference:**   [Spec](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/PowerSourceCluster.adoc#attributes) , [Test-plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/powersource.adoc#tc-ps-2-1-attributes-with-server-as-dut) .

**Logs:**

> [1650535089.588787][29787:29792] CHIP:EM: Removed CHIP MessageCounter:14107679 from RetransTable on exchange 5386i
[1650535089.588912][29787:29792] CHIP:DMG: ReportDataMessage =
[1650535089.588977][29787:29792] CHIP:DMG: {
[1650535089.589030][29787:29792] CHIP:DMG: 	AttributeReportIBs =
[1650535089.589119][29787:29792] CHIP:DMG: 	[
[1650535089.589179][29787:29792] CHIP:DMG: 		AttributeReportIB =
[1650535089.589270][29787:29792] CHIP:DMG: 		{
[1650535089.589346][29787:29792] CHIP:DMG: 			AttributeStatusIB =
[1650535089.589425][29787:29792] CHIP:DMG: 			{
[1650535089.589501][29787:29792] CHIP:DMG: 				AttributePathIB =
[1650535089.589590][29787:29792] CHIP:DMG: 				{
[1650535089.589672][29787:29792] CHIP:DMG: 					Endpoint = 0x1,
[1650535089.589760][29787:29792] CHIP:DMG: 					Cluster = 0x2f,
[1650535089.589847][29787:29792] CHIP:DMG: 					Attribute = 0x0000_0006,
[1650535089.589929][29787:29792] CHIP:DMG: 				}
[1650535089.590027][29787:29792] CHIP:DMG: 					
[1650535089.590114][29787:29792] CHIP:DMG: 				StatusIB =
[1650535089.590195][29787:29792] CHIP:DMG: 				{
[1650535089.590278][29787:29792] CHIP:DMG: 					status = 0x86 (UNSUPPORTED_ATTRIBUTE),
[1650535089.590365][29787:29792] CHIP:DMG: 				},
[1650535089.590447][29787:29792] CHIP:DMG: 				
[1650535089.590521][29787:29792] CHIP:DMG: 			},
[1650535089.590605][29787:29792] CHIP:DMG: 			
[1650535089.590672][29787:29792] CHIP:DMG: 		},
[1650535089.590757][29787:29792] CHIP:DMG: 		
[1650535089.590820][29787:29792] CHIP:DMG: 	],
[1650535089.590897][29787:29792] CHIP:DMG: 	
[1650535089.590957][29787:29792] CHIP:DMG: 	SuppressResponse = true, 
[1650535089.591025][29787:29792] CHIP:DMG: 	InteractionModelRevision = 1
[1650535089.591088][29787:29792] CHIP:DMG: }
[1650535089.591326][29787:29792] CHIP:TOO: Response Failure: IM Error 0x00000586: General error: 0x86 (UNSUPPORTED_ATTRIBUTE)
[1650535089.591480][29787:29792] CHIP:EM: Sending Standalone Ack for MessageCounter:9731713 on exchange 5386i

cjandhyala, 2022-05-25:
@Survensa is this still an issue?

Survensa, 2022-05-25:
In recent commit its fixed and the feature map value is getting as 2 and its supporting BAT mandatory attributes.so i am closing the issue.

closed 2022-05-25 by Survensa