project-chip/connectedhomeip #17596: [SQA]-THREAD In (TC-LO-2.1) Supported-locals has no entries.

Survensa, 2022-04-21:
**Problem :**
While executing the TC-LO-2.1 in Thread platform **Supported-locals** has no entries, But while executing in the RPI platform it throws the output as 8 entries.

**Expected behavior:**   SupportedLocales: 8 entries
[1650539057.043007][2885:2890] CHIP:TOO:     [1]: en-US
[1650539057.043038][2885:2890] CHIP:TOO:     [2]: de-DE
[1650539057.043067][2885:2890] CHIP:TOO:     [3]: fr-FR
[1650539057.043095][2885:2890] CHIP:TOO:     [4]: en-GB
[1650539057.043124][2885:2890] CHIP:TOO:     [5]: es-ES
[1650539057.043152][2885:2890] CHIP:TOO:     [6]: zh-CN
[1650539057.043181][2885:2890] CHIP:TOO:     [7]: it-IT
[1650539057.043209][2885:2890] CHIP:TOO:     [8]: ja-JP

**Actual behavior:**  SupportedLocales: 0 entries

**Steps to reproduce:**  `./chip-tool localizationconfiguration read supported-locales 33 0`

**Platform:** Thread

**CommitID** used:  20e1a4a705770d5222a8ebf04e12ab2ce0304df6

**Reference:**  [Spec](https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/service_device_management/LocalizationConfiguration.adoc#supportedlocales-attribute) , [Test-plan](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/Localization.adoc#tc-lo-2-1localization-configuration-cluster-attributes)

**Logs:**

**In RPI:**

> [1650539057.042853][2885:2890] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_002B Attribute 0x0000_0002 DataVersion: 2961673245
[1650539057.042945][2885:2890] CHIP:TOO:   SupportedLocales: 8 entries
[1650539057.043007][2885:2890] CHIP:TOO:     [1]: en-US
[1650539057.043038][2885:2890] CHIP:TOO:     [2]: de-DE
[1650539057.043067][2885:2890] CHIP:TOO:     [3]: fr-FR
[1650539057.043095][2885:2890] CHIP:TOO:     [4]: en-GB
[1650539057.043124][2885:2890] CHIP:TOO:     [5]: es-ES
[1650539057.043152][2885:2890] CHIP:TOO:     [6]: zh-CN
[1650539057.043181][2885:2890] CHIP:TOO:     [7]: it-IT
[1650539057.043209][2885:2890] CHIP:TOO:     [8]: ja-JP
[1650539057.043330][2885:2890] CHIP:EM: Sending Standalone Ack for MessageCounter:11868231 on exchange 15546i

**In Thread:**

> [1650538116.572188][29974:29979] CHIP:EM: Removed CHIP MessageCounter:9183892 from RetransTable on exchange 1536i
[1650538116.572312][29974:29979] CHIP:DMG: ReportDataMessage =
[1650538116.572377][29974:29979] CHIP:DMG: {
[1650538116.572431][29974:29979] CHIP:DMG: 	AttributeReportIBs =
[1650538116.572513][29974:29979] CHIP:DMG: 	[
[1650538116.572574][29974:29979] CHIP:DMG: 		AttributeReportIB =
[1650538116.572657][29974:29979] CHIP:DMG: 		{
[1650538116.572726][29974:29979] CHIP:DMG: 			AttributeDataIB =
[1650538116.572789][29974:29979] CHIP:DMG: 			{
[1650538116.572875][29974:29979] CHIP:DMG: 				DataVersion = 0x7a5e82db,
[1650538116.572947][29974:29979] CHIP:DMG: 				AttributePathIB =
[1650538116.573020][29974:29979] CHIP:DMG: 				{
[1650538116.573108][29974:29979] CHIP:DMG: 					Endpoint = 0x0,
[1650538116.573195][29974:29979] CHIP:DMG: 					Cluster = 0x2b,
[1650538116.573273][29974:29979] CHIP:DMG: 					Attribute = 0x0000_0002,
[1650538116.573354][29974:29979] CHIP:DMG: 				}
[1650538116.573432][29974:29979] CHIP:DMG: 					
[1650538116.573512][29974:29979] CHIP:DMG: 					Data = [
[1650538116.573595][29974:29979] CHIP:DMG: 						
[1650538116.573687][29974:29979] CHIP:DMG: 					],
[1650538116.573767][29974:29979] CHIP:DMG: 			},
[1650538116.573847][29974:29979] CHIP:DMG: 			
[1650538116.574039][29974:29979] CHIP:DMG: 		},
[1650538116.574125][29974:29979] CHIP:DMG: 		
[1650538116.574184][29974:29979] CHIP:DMG: 	],
[1650538116.574258][29974:29979] CHIP:DMG: 	
[1650538116.574318][29974:29979] CHIP:DMG: 	SuppressResponse = true, 
[1650538116.574381][29974:29979] CHIP:DMG: 	InteractionModelRevision = 1
[1650538116.574452][29974:29979] CHIP:DMG: }
[1650538116.574803][29974:29979] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_002B Attribute 0x0000_0002 DataVersion: 2053014235
[1650538116.574980][29974:29979] CHIP:TOO:   SupportedLocales: 0 entries
[1650538116.575210][29974:29979] CHIP:EM: Sending Standalone Ack for MessageCounter:10443902 on exchange 1536i

vivien-apple, 2022-05-18:
It sounds like a platform specific issue where the platform needs to declare which locales it supports. Currently Linux is returning an hardcoded list of locales at https://github.com/project-chip/connectedhomeip/blob/4967c0e0a51e3e0f0216c551d35b4cbde5aea7e4/src/platform/Linux/DeviceInfoProviderImpl.cpp#L222

woody-apple, 2022-05-19:
Cert Blocker Review: Assigning to @woody-apple to find a platform owner for this nRF platform error.

andy31415, 2022-05-20:
Assigning to myself as well: I have a nrf device so if this is reasonably straight forward to implement, may as well do it.

andy31415, 2022-05-20:
This is the test script that I am using to validate things:

```sh
#!/usr/bin/env bash

set -ex

function finish {
  echo "STATUS: $?"
}

trap finish EXIT

export NODEID=$RANDOM

export THREAD_CRED=hex:... # my test thread credentials here
export CHIP_TOOL="./out/linux-x64-chip-tool-ipv6only/chip-tool"

log_file="run_log_$(date +'%Y-%m-%d_%H-%M-%S').txt"
rm /tmp/chip_*

echo "NODE ID USED: $NODEID" | tee $log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL pairing ble-thread $NODEID $THREAD_CRED 20202021 3840 | tee -a $log_file
$CHIP_TOOL pairing ble-thread $NODEID $THREAD_CRED 20202021 3840  >>$log_file

echo "=============================================================" >>$log_file
echo $CHIP_TOOL localizationconfiguration read supported-locales 33 0 | tee -a $log_file
$CHIP_TOOL localizationconfiguration read supported-locales $NODEID 0 | tee -a $log_file | grep 'CHIP:DMG:'
```

PR #18687 (2022-05-25): Move DeviceInfoProvider to common code and use it in linux and NRF.
#### Problem
#17596 reports that DeviceInfoProvider is not supported in nrfconnect (specifically supported locales list is empty).

This fixes #17596.

#### Change overview
Move DeviceInfoProvider implementation as an example provider and make use of it in linux and nrf.

Tried to preserve existing behavior and have linux (and darwin app main as a sideffect) support deviceinfo provider. Ameba uses a custom build script that is out of the tree, so I could not update that one).

#### Testing
Ran

```sh
#!/usr/bin/env bash

set - [... cut]
closed 2022-05-25 by tcarmelveilleux