project-chip/connectedhomeip #17608: [TC-MF-1.9] and [TC-MF-1.10]  passing wrong payload its stopping test execution

manjunathambiger23, 2022-04-21:
Summary: 
As per Testplan of TC-MF-1.9 and TC-MF-1.10 we are attempting pass wrong payload 20 times with the response as FAILURE but its stopping the execution showing that "***** Test Failure: ../../examples/chip-tool/third_party/connectedhomeip/src/setup_payload/Base38Decode.cpp:155: CHIP Error 0x0000002F: Invalid argument" when executed.

YAML screenshot:
[image]
[image]

Testplan screenshot:
[image]

Execution logs and screenshot:
[image]
[faillog.txt](https://github.com/project-chip/connectedhomeip/files/8532509/faillog.txt)

manjunath-grl, 2022-05-10:
This issue was created by me using a account that i had to delete due to access issues, I will continue to follow these issues with my current id.

krypton36, 2022-05-12:
This appears to be an issue with the QR code. The test indicates to use an incorrect payload not an invalid one. To generate an incorrect QR code please follow the steps below:
1. From connectedhomeip root directory run the following commands
2. Run `gn gen out/host`
3. Run `ninja -v -C out/host qrcodetool`
4. Download [qrcodetool_setup_payload.txt](https://github.com/project-chip/connectedhomeip/files/8680990/qrcodetool_setup_payload.txt) 
5. Update the values to incorrect one in **qrcodetool_setup_payload.txt**
6. Run `./out/host/obj/src/qrcodetool/bin/qrcodetool generate-qr-code -f qrcodetool_setup_payload.txt`
7. Use qr code listed in command output.

bzbarsky-apple, 2022-05-28:
@manjunath-grl @cjandhyala IS there still an issue here?

manjunath-grl, 2022-05-30:
@bzbarsky-apple its resolved can you please close it