project-chip/connectedhomeip #17627: [SQA]-THREAD SetRegulatoryConfig command responds as 0 (Success) value instead 1 ( 'Error - ValueOutsideRange' )value .

Survensa, 2022-04-22:
**Problem :**

As per Test-Plan, " SetRegulatoryConfigResponse command response from DUT to TH1 with field ErrorCode as 'Error - ValueOutsideRange' and DebugText as a 'string' value". But  when TH1 sends SetRegulatoryConfig command to the DUT with NewRegulatoryConfig data value as '3', Breadcrumb as 0 and CountryCode is of type 'string'. But we are getting response as success with the value 0

**Expected  behavior:** TH should send error code as ValueOutsideRange(1)

**Actual behavior:** TH responds with error code success(0)

**Steps to reproduce:** `./chip-tool generalcommissioning set-regulatory-config 3 te9 0 1 0`

**Platform:** Thread

**Commitid** used: 20e1a4a705770d5222a8ebf04e12ab2ce0304df6

[**Reference** ](https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/General_Commissioning.adoc#tc-gc-2-4-verification-for-errorcode-message-dut-commissionee).

**logs:** 

> [1650618632.896177][3279:3284] CHIP:EM: Removed CHIP MessageCounter:48910 from RetransTable on exchange 23586i
[1650618632.896246][3279:3284] CHIP:DMG: ICR moving to [ResponseRe]
[1650618632.896354][3279:3284] CHIP:DMG: InvokeResponseMessage =
[1650618632.896413][3279:3284] CHIP:DMG: {
[1650618632.896469][3279:3284] CHIP:DMG: 	suppressResponse = false, 
[1650618632.896530][3279:3284] CHIP:DMG: 	InvokeResponseIBs =
[1650618632.896607][3279:3284] CHIP:DMG: 	[
[1650618632.896666][3279:3284] CHIP:DMG: 		InvokeResponseIB =
[1650618632.896751][3279:3284] CHIP:DMG: 		{
[1650618632.896816][3279:3284] CHIP:DMG: 			CommandDataIB =
[1650618632.896888][3279:3284] CHIP:DMG: 			{
[1650618632.896959][3279:3284] CHIP:DMG: 				CommandPathIB =
[1650618632.897047][3279:3284] CHIP:DMG: 				{
[1650618632.897131][3279:3284] CHIP:DMG: 					EndpointId = 0x0,
[1650618632.897210][3279:3284] CHIP:DMG: 					ClusterId = 0x30,
[1650618632.897303][3279:3284] CHIP:DMG: 					CommandId = 0x3,
[1650618632.897384][3279:3284] CHIP:DMG: 				},
[1650618632.897470][3279:3284] CHIP:DMG: 				
[1650618632.897544][3279:3284] CHIP:DMG: 				CommandData = 
[1650618632.897625][3279:3284] CHIP:DMG: 				{
[1650618632.897708][3279:3284] CHIP:DMG: 					0x0 = 0, 
[1650618632.897805][3279:3284] CHIP:DMG: 					0x1 = "", 
[1650618632.897890][3279:3284] CHIP:DMG: 				},
[1650618632.897959][3279:3284] CHIP:DMG: 			},
[1650618632.898042][3279:3284] CHIP:DMG: 			
[1650618632.898109][3279:3284] CHIP:DMG: 		},
[1650618632.898288][3279:3284] CHIP:DMG: 		
[1650618632.898353][3279:3284] CHIP:DMG: 	],
[1650618632.898430][3279:3284] CHIP:DMG: 	
[1650618632.898488][3279:3284] CHIP:DMG: 	InteractionModelRevision = 1
[1650618632.898545][3279:3284] CHIP:DMG: },
[1650618632.898687][3279:3284] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_0030 Command=0x0000_0003
[1650618632.898796][3279:3284] CHIP:TOO: Endpoint: 0 Cluster: 0x0000_0030 Command 0x0000_0003
[1650618632.898927][3279:3284] CHIP:TOO:   SetRegulatoryConfigResponse: {
[1650618632.899031][3279:3284] CHIP:TOO:     errorCode: 0
[1650618632.899090][3279:3284] CHIP:TOO:     debugText: 
[1650618632.899147][3279:3284] CHIP:TOO:    }
[1650618632.899234][3279:3284] CHIP:DMG: ICR moving to [AwaitingDe]
[1650618632.899350][3279:3284] CHIP:EM: Sending Standalone Ack for MessageCounter:1757833 on exchange 23586i

andy31415, 2022-04-25:
Seems very similar to #17319

Survensa, 2022-05-09:
In latest commit 11587f5bf0bb9e621edd4c867c0bc942d804d1ce , Getting the Expected outcome.

closed 2022-05-09 by Survensa