project-chip/connectedhomeip #17633: [TC-LO-1.1]Localization Configuration- Global attribute "Attributelist" has the wrong Attribute-ID.

Rajashreekalmane, 2022-04-22:
Problem:
In TC-LO-1.1 the Attribute-list Command throws the values 1&2 as the output. But as per spec the Localization Configuration Cluster, Attribute List has the 0 (ActiveLocale) & 1 (SupportedLocales).

 *Expected behavior: AttributeList: 6 entries
[1650536674.536020][29894:29899] CHIP:TOO: [1]: 0
[1650536674.536085][29894:29899] CHIP:TOO: [2]: 1
[1650536674.536178][29894:29899] CHIP:TOO: [3]: 65528
[1650536674.536241][29894:29899] CHIP:TOO: [4]: 65529
[1650536674.536324][29894:29899] CHIP:TOO: [5]: 65531
[1650536674.536386][29894:29899] CHIP:TOO: [6]: 65533
 
*Actual behavior: AttributeList: 6 entries
[1650536674.536020][29894:29899] CHIP:TOO: [1]: 1
[1650536674.536085][29894:29899] CHIP:TOO: [2]: 2
[1650536674.536178][29894:29899] CHIP:TOO: [3]: 65528
[1650536674.536241][29894:29899] CHIP:TOO: [4]: 65529
[1650536674.536324][29894:29899] CHIP:TOO: [5]: 65531
[1650536674.536386][29894:29899] CHIP:TOO: [6]: 65533
 
*Steps to reproduce: ./chip-tool localizationconfiguration read attribute-list 33 0

 * system configuration:-
 system configuration
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 2982c0c
Dut: All-Clusters-app

Spec Screenshot:-
[image]

PFA log below:-
[TC-LO-1.1.txt](https://github.com/project-chip/connectedhomeip/files/8540280/TC-LO-1.1.txt)

andy31415, 2022-05-13:
Seems the same as #18148

sowmyassp, 2022-05-13:
@Both are not same this issue is related to  mandatory global attribute ""attribute-list"" of  Localization cluster where as in issue 18148 its about how to check the set-regularity configuration of commissioning flow in device discovery cluster

vivien-apple, 2022-05-18:
It just seems that the attributes ids are wrong if in the xml definition of the cluster :/

jmeg-sfy, 2022-05-18:
seen this at TE9 breaking 2 test steps thank Vivien for doing this

PR #18561 (2022-05-18): Localization configuration correct attribute ids
#### Problem

The attribute ids per spec are `0` and `1` while the current xml says `1` and `2`.

(This PR is on top of #18559, please ignore the first commit)

Fix #17633

#### Change overview
 * Update the attribute ids

#### Testing
`./out/debug/standalone/chip-tool localizationconfiguration read attribute-list 33 0`
closed 2022-05-18 by andy31415