project-chip/connectedhomeip #17635: [TC-IDM-7.1] While verifying the report data messages for subscriptions requested from devices on different fabrics but currently DUT is responding with Same fabrics.

Rajashreekalmane, 2022-04-22:
#### Problem
As per Test-Plan, [TC-IDM-7.1] Multi Fabric Subscription DUT as the server Test-Cases The purpose of this test case
 is to verifys the report data messages for subscriptions requested from devices on different fabrics are handled 
correctly by the DUT"  but currently DUT is responding with Same fabrics".

system configuration
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: https://github.com/project-chip/connectedhomeip/commit/2982c0ca0623773327def4cc7180d07f6b959c42
Dut: All-Clusters-app
 

Steps to reproduce:-
DUT side : sudo rm -rf /tmp/chip_*
sudo ./chip-all-clusters-app --wifi

TH Side:-
./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840 (on TH1)
./chip-tool pairing open-commissioning-window 1 1 100 1000 3840 / ./chip-tool administratorcommissioning open-basic-commissioning-window 500 1 0 --timedInteractionTimeoutMs 1000 (on TH1)
./chip-tool pairing manualcode 2 35484132896 / ./chip-tool pairing onnetwork 2 20202021 (on TH2)
./chip-tool pairing open-commissioning-window 3 1 100 1000 3840 / ./chip-tool administratorcommissioning open-basic-commissioning-window 500 3 0 --timedInteractionTimeoutMs 1000(on TH3)
./chip-tool pairing manualcode 3 35484132896 / ./chip-tool pairing onnetwork 2 20202021 (on TH3)
./chip-tool operationalcredentials read fabrics 2 0 --fabric-filtered 0
./chip-tool operationalcredentials remove-fabric 2 2 0

 
PFA log:-
[cntrl1.txt](https://github.com/project-chip/connectedhomeip/files/8540555/cntrl1.txt)
[cntrl2 (1).txt](https://github.com/project-chip/connectedhomeip/files/8540556/cntrl2.1.txt)
[cntrl3.txt](https://github.com/project-chip/connectedhomeip/files/8540559/cntrl3.txt)

Test-Plan Reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/interactiondatamodel.adoc#tc-idm-7-1-multi-fabric-subscription-test-cases-dut-as-the-server

vivien-apple, 2022-04-25:
Looking at the commands it seems the the DUT is paired 3 times with different node ids but on the same fabric ?

Can you try using `--commissioner-name [alpha|beta|gamma]` feature ?

Rajashreekalmane, 2022-05-05:
@vivien-apple after tried this ./chip-tool operationalcredentials read fabrics 1 0 --fabric-filtered 0 --commissioner-name alpha
 **DUT is responding with Same fabrics**

PFA logs:-
[cntrl1 (1).txt](https://github.com/project-chip/connectedhomeip/files/8632754/cntrl1.1.txt)
[cntrl2 (2).txt](https://github.com/project-chip/connectedhomeip/files/8632757/cntrl2.2.txt)
[cntrl3 (1).txt](https://github.com/project-chip/connectedhomeip/files/8632762/cntrl3.1.txt)

erjiaqing, 2022-05-10:
I cannot reproduce this with Repl:

```
In [1]: devCtrl.CommissionIP(b'127.0.0.1', 20202021, 1)
Established CASE with Device
Commissioning complete
Out[1]: True

In [2]: await devCtrl.ReadAttribute(1, [(Clusters.OperationalCredentials.Attribu
   ...: tes.Fabrics)], fabricFiltered=True)
Out[2]: 
{
│   0: {
│   │   <class 'types.OperationalCredentials'>: {
│   │   │   <class 'chip.clusters.Attribute.DataVersion'>: 3021222368,
│   │   │   <class 'types.fabrics'>: [
│   │   │   │   FabricDescriptor(
│   │   │   │   │   rootPublicKey=b'\x04\xa1\xc2\x9b\x9eQ\x8c\x8d$!e$\xc5]R\x9e\xa6-J\xc1\xe6\xaf\xa9D\xff\xf2\\\x01g\xc4\xd82.\xf7*C:\x19/\x19E\xd9V\xd3V\x17!\xff\xe0-\xec\x92j\x12\x04cG\xf8\x19D\\\x8a\t\x809',
│   │   │   │   │   vendorId=43776,
│   │   │   │   │   fabricId=1,
│   │   │   │   │   nodeId=1,
│   │   │   │   │   label='',
│   │   │   │   │   fabricIndex=1
│   │   │   │   )
│   │   │   ]
│   │   }
│   }
}

In [3]: import chip.FabricAdmin as FabricAdmin

In [5]: fabric2 = FabricAdmin.FabricAdmin(fabricId = 2, fabricIndex = 2)
New FabricAdmin: FabricId: 2(2)

In [6]: devCtrl2 = fabric2.NewController()
Allocating new controller with FabricId: 2(2), NodeId: 1

In [9]: await devCtrl.SendCommand(1, 0, Clusters.AdministratorCommissioning.Comm
   ...: ands.OpenBasicCommissioningWindow(600), timedRequestTimeoutMs=1000)
   
In [10]: devCtrl2.CommissionIP(b'127.0.0.1', 20202021, 2)
Established CASE with Device
Commissioning complete
Out[10]: True

In [11]: await devCtrl.ReadAttribute(1, [(Clusters.OperationalCredentials.Attrib
    ...: utes.Fabrics)], fabricFiltered=True)
Out[11]: 
{
│   0: {
│   │   <class 'types.OperationalCredentials'>: {
│   │   │   <class 'chip.clusters.Attribute.DataVersion'>: 3021222371,
│   │   │   <class 'types.fabrics'>: [
│   │   │   │   FabricDescriptor(
│   │   │   │   │   rootPublicKey=b'\x04\xa1\xc2\x9b\x9eQ\x8c\x8d$!e$\xc5]R\x9e\xa6-J\xc1\xe6\xaf\xa9D\xff\xf2\\\x01g\xc4\xd82.\xf7*C:\x19/\x19E\xd9V\xd3V\x17!\xff\xe0-\xec\x92j\x12\x04cG\xf8\x19D\\\x8a\t\x809',
│   │   │   │   │   vendorId=43776,
│   │   │   │   │   fabricId=1,
│   │   │   │   │   nodeId=1,
│   │   │   │   │   label='',
│   │   │   │   │   fabricIndex=1
│   │   │   │   )
│   │   │   ]
│   │   }
│   }
}

In [12]: await devCtrl2.ReadAttribute(2, [(Clusters.OperationalCredentials.Attri
    ...: butes.Fabrics)], fabricFiltered=True)
Out[13]: 
{
│   0: {
│   │   <class 'types.OperationalCredentials'>: {
│   │   │   <class 'chip.clusters.Attribute.DataVersion'>: 3021222371,
│   │   │   <class 'types.fabrics'>: [
│   │   │   │   FabricDescriptor(
│   │   │   │   │   rootPublicKey=b'\x04\x9d>\x0c\x01R\xb7p\xacTwC\xb8\xb7yN\xef\xd0y\x18\xd2\xa6DU\xbaX\xa5\xc6\xf2\xe1o\xc3\x8d\x15\xd5tl\x1b\x9d\xefPr\xf1\x9a\xc8\xfd\x185\xac\x07\'~\x93O\xe3"\xb6nv]\xa1\x7fY\xcb\x19',
│   │   │   │   │   vendorId=43776,
│   │   │   │   │   fabricId=2,
│   │   │   │   │   nodeId=2,
│   │   │   │   │   label='',
│   │   │   │   │   fabricIndex=2
│   │   │   │   )
│   │   │   ]
│   │   }
│   }
}

In [14]: await devCtrl.ReadAttribute(1, [(Clusters.OperationalCredentials.Attrib
    ...: utes.Fabrics)], fabricFiltered=True)
Out[14]: 
{
│   0: {
│   │   <class 'types.OperationalCredentials'>: {
│   │   │   <class 'chip.clusters.Attribute.DataVersion'>: 3021222371,
│   │   │   <class 'types.fabrics'>: [
│   │   │   │   FabricDescriptor(
│   │   │   │   │   rootPublicKey=b'\x04\xa1\xc2\x9b\x9eQ\x8c\x8d$!e$\xc5]R\x9e\xa6-J\xc1\xe6\xaf\xa9D\xff\xf2\\\x01g\xc4\xd82.\xf7*C:\x19/\x19E\xd9V\xd3V\x17!\xff\xe0-\xec\x92j\x12\x04cG\xf8\x19D\\\x8a\t\x809',
│   │   │   │   │   vendorId=43776,
│   │   │   │   │   fabricId=1,
│   │   │   │   │   nodeId=1,
│   │   │   │   │   label='',
│   │   │   │   │   fabricIndex=1
│   │   │   │   )
│   │   │   ]
│   │   }
│   }
}
```

After checking the log, I found the server did generated data from different fabric.

```
[1651756088.569329][4925:4930] CHIP:TOO:   Fabrics: 1 entries
[1651756088.569447][4925:4930] CHIP:TOO:     [1]: {
[1651756088.569516][4925:4930] CHIP:TOO:       RootPublicKey: 0400D3D4041D22DC47DCF375B6145A459DDB8929B7EC4D5A40D2665F2BC7E5F4AD9790D9922249092AE7EFE391F7CD285C68D6C61B2BA2D05B2CE00D295104A71C
[1651756088.569602][4925:4930] CHIP:TOO:       VendorId: 65521
[1651756088.569658][4925:4930] CHIP:TOO:       FabricId: 1
[1651756088.569711][4925:4930] CHIP:TOO:       NodeId: 1
[1651756088.569763][4925:4930] CHIP:TOO:       Label: 
[1651756088.569815][4925:4930] CHIP:TOO:       FabricIndex: 1
[1651756088.569862][4925:4930] CHIP:TOO:      }
```

```
[1651756109.098434][4496:4501] CHIP:TOO:   Fabrics: 1 entries
[1651756109.098554][4496:4501] CHIP:TOO:     [1]: {
[1651756109.098625][4496:4501] CHIP:TOO:       RootPublicKey: 047C84328F07E539664C599E50C8656F113E8D3DF22E08A333CE0BEF947591D98C605319123A9696ABFD1ABE4C2EC681861BC9F54B46B43C370E2E1D06798F3DE7
[1651756109.098712][4496:4501] CHIP:TOO:       VendorId: 65521
[1651756109.098768][4496:4501] CHIP:TOO:       FabricId: 1
[1651756109.098820][4496:4501] CHIP:TOO:       NodeId: 2
[1651756109.098871][4496:4501] CHIP:TOO:       Label: 
[1651756109.098923][4496:4501] CHIP:TOO:       FabricIndex: 2
[1651756109.098975][4496:4501] CHIP:TOO:      }
```

(Please check the fabric index and the root public key, they are different)

However, the two fabric comes with the same fabric id.

erjiaqing, 2022-05-10:
I would suggest this is a issue related to **OpCred / Commissioning** rather than IDM, since the fabric filter feature should be working according to the output (different fabric index, different root public key from different fabric)

vivien-apple, 2022-05-18:
Sorry my comment was not clear enough.

The initial set of commands should be:
```
./chip-tool pairing ble-wifi 1 zigbeehome matter123 20202021 3840 (on TH1)
./chip-tool pairing open-commissioning-window 1 1 100 1000 3840 / ./chip-tool administratorcommissioning open-basic-commissioning-window 500 1 0 --timedInteractionTimeoutMs 1000 (on TH1)

./chip-tool pairing manualcode 2 35484132896 / ./chip-tool pairing onnetwork 2 20202021 --commissioner-name beta (on TH2)
./chip-tool pairing open-commissioning-window 3 1 100 1000 3840 / ./chip-tool administratorcommissioning open-basic-commissioning-window 500 3 0 --timedInteractionTimeoutMs 1000 --commissioner-name beta (on TH2)

./chip-tool pairing manualcode 3 35484132896 / ./chip-tool pairing onnetwork 2 20202021 --commissioner-name gamma (on TH3)
./chip-tool operationalcredentials read fabrics 2 0 --fabric-filtered 0 --commissioner-name gamma
./chip-tool operationalcredentials remove-fabric 2 2 0 --commissioner-name gamma
```

If you don't pass `commissioner-name` to commands you end up using the "default" fabric (which is `alpha`)

woody-apple, 2022-05-19:
Cert Blocker Review: Per @cjandhyala, this can be closed.

closed 2022-05-19 by woody-apple

bzbarsky-apple, 2022-05-19:
Note that the original steps listed in this issue will fail on tip for two reasons:

1. The commissioning window timeouts are too short (need to be at least 180)
2. If that's fixed, we follow the spec now and disallow commissioning a device with multiple node ids onto the same fabric:
    ```
    [1652979019782] [29245:51504946] CHIP: [TOO] Run command failure: ../../../examples/chip-tool/third_party/connectedhomeip/src/controller/CHIPDeviceController.cpp:1216: CHIP Error 0x0000007E: Trying to add a NOC for a fabric that already exists
    ```