project-chip/connectedhomeip #17640: iOS CHIPTool: scanNetworks API is failing

ajay-gantayet, 2022-04-22:
#### Problem
I am using the following iOS CHIP API to scan networks available to the matter device after initiating device pairing.

```
- (void)scanNetworksWithParams:(CHIPNetworkCommissioningClusterScanNetworksParams * _Nullable)params
             completionHandler:(void (^)(CHIPNetworkCommissioningClusterScanNetworksResponseParams * _Nullable data,
                                   NSError * _Nullable error))completionHandler

```
For the params I am passing nil in order to get the list of wifis available. But I am getting the following error:
**The cluster command is malformed, has missing fields, or fields with invalid values. Command not carried out.**
connectedhomeip commit ID: **6109d14348023229614227775b57b8b4c80f131a**

 * Espected behavior: API should return an object with the list of networks available to the matter device
 * actual behavior: Gives the following error "The cluster command is malformed, has missing fields, or fields with invalid values. Command not carried out."
 * steps to reproduce: Invoking this API after device is successfully paired.

dhrishi, 2022-04-22:
@bzbarsky-apple @vivien-apple Can you please help?

bzbarsky-apple, 2022-04-22:
@ajay-gantayet What is the actual payload received by the device?  

For iOS, is the framework being used also built from https://github.com/project-chip/connectedhomeip/commit/6109d14348023229614227775b57b8b4c80f131a?  Or is it some framework version from some older SHA that comes as part of iOS?

bzbarsky-apple, 2022-04-22:
Nevermind, I can reproduce this using the framework from https://github.com/project-chip/connectedhomeip/commit/6109d14348023229614227775b57b8b4c80f131a if I just hack chip-tool-darwin to send `nil` for the params struct.

What's going on here is that the "ssid" field of this command is both nullable _and_ optional.  But it's only allowed to be omitted in the Thread case; in the wifi case it must be included.

The Darwin API uses `nil` to represent both "missing" and "null", on the assumption that there are never cases when those would have different behavior, but in this specific case they _do_ in fact have different behavior: null means "scan all networks" and missing is a malformed command if the server is a wifi network commissioning cluster.

Sending `nil` for the entire params struct corresponds to using `nil` for all the fields (and is allowed because all the fields are optional _for the sender_, in that if you were sending ScanNetworks for Thread you would in fact leave out the SSID).

In the short run, this can be worked around by the API consumer sending an empty string for the SSID, which would then have a value and is defined by the spec to behave the same as sending null.  But I need to think a little about how we should handle this from a general API perspective.

ajay-gantayet, 2022-04-26:
@bzbarsky-apple  thanks for the update. 
I tried passing empty string for ssid and it worked for me.

bzbarsky-apple, 2022-04-29:
I'm leaving it open for that for now, yes.

bzbarsky-apple, 2022-12-12:
From a general API perspective, there should be no APIs that give "missing" and "null" different meanings, so we will continue to represent both with `nil` in the Darwin API, looks like.

closed 2022-12-12 by bzbarsky-apple