project-chip/connectedhomeip #17666: Facing pairing issues for ECM test of Multi Fabrics testcases

manjunathambiger23, 2022-04-23:
Summary:
Facing Pairing issue for ECM tests. if commissioning window is opened for 1st Commissioner ,if we try to open commissioning process with DUT_CE for 2 nd Commissioner showing FAILURE response

Testplan screenshot:
[image]
[image]

Execution screenshot:
[image]

manjunath-grl, 2022-07-26:
Added diff: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Commissioning_error

manjunath-grl, 2022-07-26:
@raju-apple discussed with QA team they are not facing this issue and they are following the testplan no Preconditions are there and script is also as per testplan but in " step 13: TH_CR2 starts a commissioning process with DUT_CE " we are getting failure response when executed.
YAML script [TC_CADMIN_1_5](https://github.com/project-chip/connectedhomeip/blob/master/src/app/tests/suites/certification/Test_TC_CADMIN_1_5.yaml).

raju-apple, 2022-07-26:
@krypton36 can you please check this issue out when you get a chance ?

krypton36, 2022-07-27:
The PAKEVerifier and the setup PIN code do not match. 
The setup qr parses to the following (20202021):
[image]

The PAKEVerifier is using the following (47514138):
[image]

This corresponds to the error message:
`Failed to verify peer's MAC. This can happen when setup code is incorrect.`

Please make sure that the PAKEVerifier for ECM and the SetupQRCode pin code match.

manjunath-grl, 2022-07-28:
@krypton36 can you please tell the steps how to generate PAKEVerifier using setup PIN code.

krypton36, 2022-08-01:
ninja -v -C out/host spake2p
./out/host/spake2p gen-verifier -p 20202021 -i 1000 -l 32 -s 5350414B453250204B65792053616C74 -o pake_verifier.txt

krypton36, 2022-08-01:
5350414B453250204B65792053616C74 might need to be encoded to base64

krypton36, 2022-08-01:
echo "5350414B453250204B65792053616C74" | xxd -r -p | base64

krypton36, 2022-08-01:
Or `echo -n "SPAKE2P Key Salt" | od -A n -t x1 | xxd -r -p | base64`

manjunath-grl, 2022-08-10:
@raju-apple can you please close the ticket its resolved.

closed 2023-02-19 by stale[bot]