project-chip/connectedhomeip #17684: src/platform platform directory support Secondary development

lbit20131014-berry, 2022-04-25:
#### Problem
The platform-related files in the src/platform directory are often modified by the manufacturer, so whether the user-defined location of this directory can be supported without changing the connectedhomeip source code. For changing chip_device_platform=external, but the variable chip_device_platform is used in many directories. If it is changed in this way, it will lead to the need to know all the places where the variable chip_device_platform appears. It is also very workload for the merge after the subsequent connectedhomeip update.
There is a _platform_target variable in the src/platform/BUILD.gn file, and this variable is only used by this file, so is it possible to set this variable through gn build without changing the variable chip_device_platform to achieve application user platform related code

 #### Proposed Solution
<suggested fix, suggested enhancement>

chshu, 2022-04-25:
It's currently supported in ESP platform, via a simply menuconfig change:

https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L789

lbit20131014-berry, 2022-04-25:
> It's currently supported in ESP platform, via a simply menuconfig change:
> 
> https://github.com/project-chip/connectedhomeip/blob/master/config/esp32/components/chip/Kconfig#L789

@chshu Thanks Reply
If this option is configured, the following results will occur:
chip_device_platform=external
chip_platform_target=CONFIG_CHIP_EXTERNAL_PLATFORM_TARGET
At this time, for example, the chip_device_platform in the src/credentials/BUILD.gn file is not esp32, and is it necessary to provide the esp32 configuration elsewhere? This is just one of the directories, and there are some corresponding to the chip_device_platform directory. , for the subsequent connectedhomeip update, the user needs to retrieve all the files involving chip_device_platform, which will be a very troublesome thing, I don't know if I understand it right

gjc13, 2022-04-26:
@lbit20131014-berry The external platform shall provide a custom build target in its `BUILD.gn`. The new target will replace the one in `src/platform/BUILD.gn`. Let's say we've added a custom platform at folder `esp32_custom`. The folder shall contain file:
```
esp32_custom
├── BLEManagerImpl.h
├── BUILD.gn
.....
```
The BUILD.gn will define target `esp32_custom`:
```
static_library("esp32_custom") {
  sources = [
...
```

Then in ESP32 Kconfig, we'll set `CONFIG_CHIP_EXTERNAL_PLATFORM_TARGET` to be the path to the `esp32_custom` folder. All the `chip_device_platform` variable in `BUILD.gn` files shall be correctly updated.

closed 2022-11-01 by stale[bot]