project-chip/connectedhomeip #17688: [TC-TSTAT-1.1] DUT fails to list the supported commands for GeneratedCommandList attribute

Rajashreekalmane, 2022-04-25:
#### Problem
As per the Test-Plan, GeneratedCommandList attribute should list the supported commands that is GetWeeklyScheduleResponse and 
GetRelayStatusLogResponse but currently DUT responds with 0 entries.

 * expected behavior:- DUT should responds with supported commands. 
 * actual behavior:- 0 entries

 * steps to reproduce:-./chip-tool thermostat read generated-command-list 1 1

 *system configuration:-
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 077e44eadf46f349916b3f4f358da6126be2ed48
Dut: Thermostat app 
 
PFA log below:-
[GeneratedCommandList.txt](https://github.com/project-chip/connectedhomeip/files/8554422/GeneratedCommandList.txt)

Test-plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/thermostat.adoc#tc-tstat-1-1-global-attributes-with-server-as-dut

bauerschwan, 2022-05-06:
@Rajashreekalmane please close if you agree.

bauerschwan, 2022-05-06:
Indeed the all-cluster-app supports SCH so GetWeeklySchedule is contained in AcceptedCommandList attribute. Therefore GetWeeklyScheduleResponse has to be contained in GeneratedCommandList attribute.

But the optional command GetRelayStatusLog is not contained in AcceptedCommandList attribute.
Therefore GetRelayStatusLogResponse shall not be contained in GeneratedCommandList attribute.

bauerschwan, 2022-05-06:
@chrisbeach59 who is responsible for this?

Rajashreekalmane, 2022-05-17:
Yes DUT (Thermostat app) support  GetWeeklySchedule which is  mandatory so it  should print the ID of it while running the "./chip-tool thermostat read generated-command-list 1 1"

bauerschwan, 2022-05-17:
For me this looks like a zap issue because the GENERATED_COMMANDS in the zap-generated file endpoint_config.h contains wrong information. At least this is the case with code based on TE9 tag.

Here as an example the ZAP UI of our DUT:
[image]

Here the GENERATED_CLUSTERS in endpoint_config.h:
[image]

Here the GENERATED_COMMANDS in endpoint_config.h:
[image]

So this is definitively wrong and should be fixed on zap side.

bzbarsky-apple, 2022-05-17:
Did this not get fixed in https://github.com/project-chip/connectedhomeip/pull/17833 ?

bzbarsky-apple, 2022-05-17:
The answer is: No, it did not.  For thermostat, we are very definitely listing response commands which are not enabled, and correspond to request commands that are also not enabled, in the "AcceptedCommandList", which is all sorts of broken.

bzbarsky-apple, 2022-05-17:
@erjiaqing

chrisbeach59, 2022-05-17:
@Rajashreekalmane only setpointraiselower is mandatory.   It is mandatory if schedule is enabled.  Scheduling is not supported in R1

bzbarsky-apple, 2022-05-19:
https://github.com/project-chip/zap/pull/515 will fix things.

closed 2022-05-20 by andy31415