project-chip/connectedhomeip #17698: Timeout error occured occasionally in the chip-tool [All-clusters-app]

somu1710, 2022-04-25:
**Problem**
Timeout error occured occasionally during the executing of test cases on raspi and thread platform . 

**Commit ID** -077e44eadf46f349916b3f4f358da6126be2ed48

**platform:** - Raspi,Thread

**Error:**

**DUT-** [1650885627.816408][11255:11255] CHIP:EM: OnMessageReceived failed, err = ../../third_party/connectedhomeip/src/transport/PairingSession.cpp:29: CHIP Error 0x0000000B: No memory

**CTRL-** [1650885657.611084][9442:9442] CHIP:DL: System Layer shutdown
[1650885657.611268][9442:9442] CHIP:TOO: Run command failure: ../../commands/common/CHIPCommand.cpp:392: CHIP Error 0x00000032: Timeout

**Actual logs :**

[new_log_dut.txt](https://github.com/project-chip/connectedhomeip/files/8556459/new_log_dut.txt)
[ctrl_log.txt](https://github.com/project-chip/connectedhomeip/files/8556462/ctrl_log.txt)

kghost, 2022-04-26:
Is any SecureSession being evicted during these testing ? Seems a duplicated issue of #17568

andy31415, 2022-04-26:
@somu1710 - do you have some reproduction steps that can be performed to debug this? the `occasionally` is vague and just looking at logs it seems hard to figure out how the session table got filled and when.

At a first glance, it looks like DUT has several sessions created towards it and eventually runs out of slots. Need to figure out who creates those sessions (likely chip-tool ... but why has it changed) and figure out a way to properly expire/use LRU for them.

bzbarsky-apple, 2022-04-26:
chip-tool sets up a new CASE session every time it's invoked.

Up until #17422 setting up a new CASE session with A as initiator and B as responder would expire all existing CASE sessions on B that had A as a peer.  But in #17422 that was removed.  Now we keep establishing sessions until the session table on the device fills up, and then things stop working.

@msandstedt @mrjerryjohns

bzbarsky-apple, 2022-04-26:
I believe we should restore the eviction logic we used to have, poor as it was, until the better logic suggested in #17568 is implemented.

andy31415, 2022-04-26:
That makes it clear to reproduce: run chip-tool more than 16 times. If that is true, we should be able to validate fixes (no more bisecting since we know what introduced this).

somu1710, 2022-04-26:
This issue is also persist in  OTA provider and requestor while executing default ota provider command.

**Error**

CHIP:SWU: Failed to connect to node 0x000000000000007B: ../../examples/ota-requestor-app/linux/third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:282: CHIP Error 0x00000032: Timeout

tcarmelveilleux, 2022-04-26:
I am looking at temporarily reimplementing CASE session discarding as before. Will report here if I get something working, before EOD Apr 26, EDT timezone (GMT -4)

Survensa, 2022-04-26:
@tcarmelveilleux 
I too faced same issue now with RPI - RPI using BLe-wifi

Steps followed :

> 1. Paired CNTRL to DUT.
> 2. Executed samlple command A :` ./chip-tool onoff read feature-map 1 1`
> 3.  Executred sample Command B :` ./chip-tool onoff read attribute-list 1 1`
> 4. Executed Sample Command C :  `./chip-tool onoff read accepted-command-list 1 1`

log Observations :

**For command A :**

`[1650997628.391520][6491:6496] CHIP:TOO:   FeatureMap: 1`

**For Command B :**

`[1650997635.720191][6497:6497] CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/protocols/secure_channel/CASESession.cpp:1508: CHIP Error 0x00000054: Invalid CASE parameter`

**For command C :**

`[1650997648.558313][6503:6503] CHIP:TOO: Run command failure: ../../third_party/connectedhomeip/src/crypto/CHIPCryptoPALOpenSSL.cpp:336: CHIP Error 0x000000AC: Internal error`

Full log for referance :

[IN VA CP.txt](https://github.com/project-chip/connectedhomeip/files/8566007/IN.VA.CP.txt)

tcarmelveilleux, 2022-04-26:
The quick-fix is up in https://github.com/project-chip/connectedhomeip/pull/17779

cjandhyala, 2022-04-26:
@tcarmelveilleux with pulling the above PR both on controller and app and rebuilding, the above issue still persists. Attached logs. 

[all_cluster_app_log.txt](https://github.com/project-chip/connectedhomeip/files/8567345/all_cluster_app_log.txt)
[chip_tool_log_raspi.txt](https://github.com/project-chip/connectedhomeip/files/8567346/chip_tool_log_raspi.txt)

PR #17779 (2022-04-27): Quick-fix running out of CASE sessions
#### Problem
- From #17422 onwards, "previous" CASE sessions were not expired
  in terms of the session cache if sessions were constantly
  re-established.
- A root cause solution will be to solve #17568, but in the meantime,
  cert testing started failing after N (where N is session manager
  pool size) runs.

Fixes #17698

#### Change overview
This PR cleans-up the previous sessions from same node (duplicate sessions
that should not happen). This is a bit unclean but will do to unblock
testing since it was good enough before, and the better method is no [... cut]
commit 1b5c108a (2022-04-27): Quick-fix running out of CASE sessions (#17779)

* Quick-fix running out of CASE sessions

- From #17422 onwards, "previous" CASE sessions were not expired
  in terms of the session cache if sessions were constantly
  re-established.
- A root cause solution will be to solve #17568, but in the meantime,
  cert testing started failing after N (where N is session manager
  pool size) runs.

This PR cleans-up the previous sessions from same node (duplicate sessions
that should not happen). This is a bit unclean but will do to unblock
testing since it was good enough before, and the better method i [... cut]
closed 2022-04-27 by kghost