project-chip/connectedhomeip #17700: Color control cluster doesn't support feature map for hue/saturation/color temperature

nivi-apple, 2022-04-25:
#### Problem
Color control cluster exposes apis for reading/writing and subscribing to Hue, Saturation and color temperature. However it doesn't support the feature map attribute. Hence it's not spec complaint. See the feature map table from the spec -

<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Strict//EN" "http://www.w3.org/TR/xhtml1/DTD/xhtml1-strict.dtd">
<html xmlns="http://www.w3.org/1999/xhtml">
	<head>
		
		<title>Connected Home over IP Application Clusters</title>
	</head>
	<body>
		<div class="page" title="Page 106">
			<div class="layoutArea">
				<div class="column">
					<p><span style="font-size: 13.000000pt; font-family: 'e7491e+NotoSerif'; font-weight: 700; color: rgb(20.000000%, 20.000000%, 20.000000%)">3.2.5. Features
</span></p>
					<p><span style="font-size: 11.000000pt; font-family: '32eac5+NotoSerif'; color: rgb(20.000000%, 20.000000%, 20.000000%)">This cluster SHALL support the FeatureMap bitmap attribute as defined below.
</span></p>
				</div>
			</div>
			

Identifier Name
--
0x0300 Color Control

		</div>
	</body>
</html>
 >

 #### Proposed Solution
Need to add support for feature map attribute.

bzbarsky-apple, 2022-05-22:
@nivi-apple I believe this was fixed in #17082.  Can you please confirm?  You filed this issue much later than that, but I don't know what SHA you were using....

nivi-apple, 2022-05-23:
@bzbarsky-apple I was on an older SHA. I will verify on the tip of tree as well next time :). closing the issue. Thanks

closed 2022-05-23 by nivi-apple