project-chip/connectedhomeip #17723: Source activate.sh error

TheWalkingDead1024, 2022-04-26:
When I try to source activate.sh to build preparation for CHIP project (I use M1 Macbook Pro), I got an error like this:

WELCOME TO...

           ░▓░
           ▓█▓
           ▓█▓                                                     ▒█     ▒█
      ▒██▒▒▓██▒███▒                ░▒▓▒░  ░▒▓▒░       ░░▓█▒░ ░█  █████████████░    ░▒█▒░       ░░▒░
       ░▓█████▓██░               ▒█▒░░▒▓██▓▒░░▒█▒   ░█▓▒░░▒████    █▓░    █▓░    ▒█▒░░░▒█▓░   ██▒░░
    ▒█▒░         ░██▒           ░█░     █▓     ░█░ ░▓▒      ░▓█    █▓     █▓    ▒█░░    ░██  ▒▓
    ░▓██▓░     ░██▓█░           ░█      ▓█      █░ ░█░       ██    █▓     █▓    ▓██████████  ▒█
     ░▓███▒   ▒███▒░            ░█      ▓█      █░  ██░    ░███    █▓     █▓    ░█▒░         ▒█
 ░▒████████░ ░███▓▓█▓▓▒         ░█      ▒▒      █░   ░█▓██▓█░▒█    ░▓▓█░  ░▓▓█░   ▒▓▓██▓█░   ▒▓
 ░██▒░  ▒██▒ ▒██░  ░▒█▓
         ░▓░ ░▓░

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (7.3s)
  Setting up Python environment.....[-]
ninja: Entering directory `/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/.environment/gn_out'
[1/223] python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_presubmit/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_presubmit/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --no-build-isolation --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_presubmit/py
FAILED: python/gen/third_party/pigweed/repo/pw_presubmit/py/py._run_pip_install.pw_pystamp
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../third_party/pigweed/repo/pw_presubmit/py --default-toolchain=//build/toolchain/host:mac_arm64_gcc --current-toolchain=//third_party/pigweed/repo/pw_build/python_toolchain:python --touch python/gen/third_party/pigweed/repo/pw_presubmit/py/py._run_pip_install.pw_pystamp --capture-output --module pip -- install --no-build-isolation --constraint ../../scripts/constraints.txt --editable ../../third_party/pigweed/repo/pw_presubmit/py
Obtaining file:///Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_presubmit/py
  Checking if build backend supports build_editable: started
  Checking if build backend supports build_editable: finished with status 'done'
  Preparing metadata (pyproject.toml): started
  Preparing metadata (pyproject.toml): finished with status 'done'
Collecting yapf==0.31.0
  Using cached yapf-0.31.0-py2.py3-none-any.whl (185 kB)
Collecting scan-build==2.0.19
  Using cached scan-build-2.0.19.tar.gz (46 kB)
  Preparing metadata (setup.py): started
  Preparing metadata (setup.py): finished with status 'done'
ERROR: Could not find a version that satisfies the requirement pw-cli (from pw-presubmit) (from versions: none)
ERROR: No matching distribution found for pw-cli
ninja: build stopped: subcommand failed.
['ninja', '-C', '/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/.environment/gn_out', '-v', ':python_packages.install']

Traceback (most recent call last):
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 286, in install_packages
    subprocess.check_call(ninja_cmd, stdout=outs, stderr=outs)
  File "/Users/luhaobo/.pyenv/versions/3.9.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 715, in <module>
    sys.exit(main())
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 707, in main
    return EnvSetup(**vars(parse())).setup()
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 419, in setup
    result = step(spin)
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 538, in virtualenv
    if not virtualenv_setup.install(
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 301, in install
    install_packages(gn_target)
  File "/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 289, in install_packages
    raise subprocess.CalledProcessError(err.returncode, err.cmd,
subprocess.CalledProcessError: Command '['ninja', '-C', '/Users/luhaobo/Documents/GitHub/connectedhomeipArm64/.environment/gn_out', '-v', ':python_packages.install']' returned non-zero exit status 1.

wickyou23, 2022-04-27:
I got the same error. Are there solutions for this?

rafaliya-amazon, 2022-04-28:
I did raise the same issue: https://github.com/project-chip/connectedhomeip/issues/17873

I wasn't sure we already had a similar issue raised.

rafaliya-amazon, 2022-04-29:
Please try to install Python 3.9.6 & then perform the following: 

      Application << Python 3.9.6 << Run "Install Certificates.command"
      
This solved the issue for me on my M1 Apple Silicon.

TheWalkingDead1024, 2022-05-07:
> Please try to install Python 3.9.6 & then perform the following:
> 
> ```
>   Application << Python 3.9.6 << Run "Install Certificates.command"
> ```
> 
> This solved the issue for me on my M1 Apple Silicon.

it is not working for me. I am using the branch interop_testing_2022_01_18. what branch are you using?

rafaliya-amazon, 2022-05-09:
> 
> it is not working for me. I am using the branch interop_testing_2022_01_18. what branch are you using?

I am using origin/master. 
I intentionally used master for the first time as there might be some scripts or any other dependencies which master have, and other branches might not have.

From the logs, your python version is [3.9.9](https://www.python.org/downloads/release/python-399/).
Please try uninstalling this python version and try installing [v3.9.6](https://www.python.org/downloads/release/python-396/)

closed 2022-10-20 by TheWalkingDead1024