project-chip/connectedhomeip #17730: ScanNetworks returns no results

ChengqiangShe, 2022-04-26:
During the network Commissioning, the app notifies the sub-device to scan the surrounding Thread network through the ScanNetworks command and returns it to the app, but the scan result is not returned. It is suspected that during the BLE connection, scanning the Thread network will cause BLE and thread conflicts, thus losing the scanning results. Eventually lead to no thread network information returned

jmartinez-silabs, 2022-04-28:
Scan network should be done once the device is commissioned and can communicate with the chip-tool. It can then Scan for thread networks in his vicinity. I might not understand what you are attempting.

ChengqiangShe, 2022-04-29:
The App wants to know whether there is a suitable Thread network near the device during the network Commissioning process, so as to give the user a prompt

ChengqiangShe, 2022-05-05:
After the device PASE is completed, the app will scan the thread network. This is a new feature added to our app

bzbarsky-apple, 2022-06-30:
@jmartinez-silabs The use case for ScanNetworks that led to it being added to the spec is to use it during commissioning to see which networks the device can see.  Doing ScanNetworks once you are on an operational network is fairly pointless....

See step 11 of commisioning flows in the spec, where it says:

> Commissioner MAY learn about the networks visible to the Commissionee using ScanNetworks command (see Section 11.8.8.2, “ScanNetworks Command”).

as part of talking about sending AddOrUpdate*Network to the commissionee.

jmartinez-silabs, 2022-07-04:
@bzbarsky-apple Yes I understand Boris. What I meant is that with the current examples and use of chip-tool you cannot use it to just do a BLE connect and send a scan thread network. You need to do the ble-thread command to commission the device and then you can send scan-network command. In that use case, it can be useful to see other networks and the area and maybe change network with the AddOrUpdate command.

However, @ChengqiangShe does seem to either have modifications in the chip-tool or his own controller. If that is the case, that can totally work but he also needs to make sure the end device is in the right thread state to scan. Currently, before any commissioning attempts the device thread status is disabled.

chrisdecenzo, 2022-07-12:
@jmartinez-silabs - it sounds like we need to add some test cases to make sure ScanNetworks can be called over the PASE connection (before AddNetworks has been called). That is the most likely/useful time to use this command.

bzbarsky-apple, 2023-01-13:
@ChengqiangShe is there still an issue left here?

closed 2023-10-15 by stale[bot]