project-chip/connectedhomeip #17737: [TC-MC-1.2]--TH fails to read all the features from DUT using feature-map attribute.

Rajashreekalmane, 2022-04-26:
#### Problem
As per Spec, TH should read all the features from DUT but currently TH fails to read all
 the features from DUT using feature-map attributes.

 * expected behavior:- TH should read 
 Bit 0 - Set to 1 if the DUT supports Navigation Keys (PICS_NV_S is true)
 Bit 1 - Set to 1 if the DUT supports Location Keys (PICS_LK_S is true)
 Bit 2 - Set to 1 if the DUT supports Number Keys (PICS_NK_S is true)
 
* actual behavior:- Bit 1 - Set to 1 if the DUT supports Location Keys (PICS_LK_S is true)

 * steps to reproduce:-sudo ./chip-tool keypadinput read feature-map 1 1

 * system configuration:-
Chip-Tool: RPI
DUT: RPI
Platform : IP and Wifi
Commit-ID: 3a7f03a0a1c58c4f6cf69ea7e6b808feff3b6636
Dut: TV-app
 
Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mc-1-3-global-attributes-application-launcher-cluster-dut-as-server

Spec Screenshot:-
[image]

PFA log below:-
[[TC-MC-1.3]Feature map.txt](https://github.com/project-chip/connectedhomeip/files/8562596/TC-MC-1.3.Feature.map.txt)

Rajashreekalmane, 2022-04-26:
Facing the same issue in channel cluster test-case[TC-MC-1.6].
expected behavior:- TH should read
Bit 0 - Set to 1 if the DUT supports Channel Lists (PICS_CL_S is true)
 Bit 1 - Set to 1 if the DUT supports Lineup Info (PICS_LI_S is true)
actual behavior:- Bit 1 - Set to 1 if the DUT supports Location Keys (PICS_LK_S is true)

Test-Plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mc-1-6-global-attributes-channel-cluster-dut-as-server

PFA log below:-
[TC-MC-1.5.txt](https://github.com/project-chip/connectedhomeip/files/8563379/TC-MC-1.5.txt)

chrisdecenzo, 2022-04-26:
Hi Lazar, can you confirm whether this is a TH issue or a tv-app issue? I know that you were working in the feature map area recently.

Rajashreekalmane, 2022-04-27:
Facing the same issue in Content launcher cluster.

 expected behavior:- TH should read
Bit 0 - Set to 1 if the DUT supports Content Search (PICS_CONTENTSEARCH is true)
Bit 1 - Set to 1 if the DUT supports URL Playback (PICS_URLPLAYBACK is true)

actual behavior:- Bit 1 - Set to 1 if the DUT supports Location Keys (PICS_LK_S is true)

Step to reproduce:-
sudo ./chip-tool  contentlauncher  read feature-map 1 3

Test-plan reference:-
https://github.com/CHIP-Specifications/chip-test-plans/blob/master/src/cluster/media.adoc#tc-mc-1-11-global-attributes-content-launcher-cluster-dut-as-server

PFA log below:-
[TC-MC-1.11.txt](https://github.com/project-chip/connectedhomeip/files/8569700/TC-MC-1.11.txt)

closed 2022-05-16 by chrisdecenzo