project-chip/connectedhomeip #17747: [TC-GC-2.3] need YAML support to send commands in  between commissioning process

manjunathambiger23, 2022-04-26:
Summary:
Need YAML support for Step5 :How do we send command in  between commissioning process

Testplan screenshot:
[image]

manjunath-grl, 2022-05-10:
This issue was created by me using a account that i had to delete due to access issues, I will continue to follow these issues with my current id.

krypton36, 2022-05-12:
@manjunath-grl this should be possible if you do not do the WaitForCommissionee command. You will have to run each of the commands to pair.

raju-apple, 2022-10-11:
@krypton36 @manjunath-grl Can you guys sync up on the next steps here please ? Creating a list for the items that we need  from the SDK team post 1.0, so want to have clear requirements for each task

raju-apple, 2022-11-01:
@manjunath-grl , were you able to sync up ?

closed 2023-05-08 by stale[bot]