project-chip/connectedhomeip #17797: Getting Segfault when sending group messaging

cjandhyala, 2022-04-27:
With ToT (13ef4a3d03f4ecdf2afd865b8c9783a856eb5801) ,  I couldnt get GroupMsging working either on Mac or linux , 

Here are the commands I used on Mac
sudo ./out/debug/chip-tool groups add-group 0x4141 Light 9 1
sudo ./out/debug/chip-tool groupkeymanagement key-set-write '{"groupKeySetID": 42,"groupKeySecurityPolicy": 0, "epochKey0":"d0d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime0": 2220000,"epochKey1":"d1d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime1": 2220001,"epochKey2":"d2d1d2d3d4d5d6d7d8d9dadbdcdddedf", "epochStartTime2": 2220002 }' 9 0
sudo ./out/debug/chip-tool groupkeymanagement write group-key-map '[{"groupId": 16705, "groupKeySetID": 42, "fabricIndex": 0}]' 9 0
On the local chip-tool I followed
4. sudo ./out/debug/chip-tool groupsettings add-group TestName3 0x4141.  --- Basically created same group on chip-tool side as well
5. sudo ./out/debug/chip-tool groupsettings add-keysets 0xAAAA 0 0x000000000021dfe0 hex:d0d1d2d3d4d5d6d7d8d9dadbdcdddedf
6. sudo ./out/debug/chip-tool groupsettings bind-keyset 0x4141 0xAAAA
Now when I try to send the on command on the group I get error, No route to host.. this when I send this command on Mac
 ```
% sudo ./out/debug/chip-tool onoff on 0xffffffffffff4141 1           
[1651025910939] [28698:9495563] CHIP: [DL] KVS will be written to: file:///tmp/chip_kvs
[1651025910952] [28698:9495563] CHIP: [DL] _Init
[1651025910952] [28698:9495563] CHIP: [IN] UDP::Init bind&listen port=0
[1651025910953] [28698:9495563] CHIP: [IN] UDP::Init bound to port=62302
[1651025910953] [28698:9495563] CHIP: [IN] UDP::Init bind&listen port=0
[1651025910953] [28698:9495563] CHIP: [IN] UDP::Init bound to port=51387
[1651025910953] [28698:9495563] CHIP: [IN] BLEBase::Init - setting/overriding transport
[1651025910953] [28698:9495563] CHIP: [IN] TransportMgr initialized
[1651025910953] [28698:9495563] CHIP: [DIS] Init fabric pairing table with server storage
[1651025910953] [28698:9495563] CHIP: [IN] Loading from storage for fabric index 0x1
[1651025910954] [28698:9495563] CHIP: [IN] Loading from storage for fabric index 0x2
[1651025910954] [28698:9495563] CHIP: [IN] Loading from storage for fabric index 0x3
[1651025910955] [28698:9495563] CHIP: [IN] Loading from storage for fabric index 0x4
[1651025910955] [28698:9495563] CHIP: [IN] Loading from storage for fabric index 0x5
[1651025910956] [28698:9495563] CHIP: [IN] Loading from storage for fabric index 0x6
[1651025910956] [28698:9495563] CHIP: [DIS] Add fabric pairing table delegate
[1651025910959] [28698:9495563] CHIP: [ZCL] Using ZAP configuration...
[1651025910959] [28698:9495563] CHIP: [CTL] System State Initialized...
[1651025910959] [28698:9495563] CHIP: [CTL] Setting attestation nonce to random value
[1651025910959] [28698:9495563] CHIP: [CTL] Setting CSR nonce to random value
[1651025910959] [28698:9495563] CHIP: [CTL] Setting attestation nonce to random value
[1651025910959] [28698:9495563] CHIP: [CTL] Setting CSR nonce to random value
[1651025910959] [28698:9495563] CHIP: [CTL] Generating NOC
[1651025910959] [28698:9495563] CHIP: [DIS] Verifying the received credentials
[1651025910960] [28698:9495563] CHIP: [DIS] Added new fabric at index: 0x4, Initialized: 1
[1651025910960] [28698:9495563] CHIP: [DIS] Assigned compressed fabric ID: 0xC664F1AF5F59144D, node ID: 0x000000000001B669
[1651025910960] [28698:9495563] CHIP: [CTL] Joined the fabric at index 4. Compressed fabric ID is: 0xC664F1AF5F59144D
[1651025910960] [28698:9495563] CHIP: [CTL] Setting attestation nonce to random value
[1651025910960] [28698:9495563] CHIP: [CTL] Setting CSR nonce to random value
[1651025910960] [28698:9495563] CHIP: [CTL] Generating NOC
[1651025910961] [28698:9495563] CHIP: [DIS] Verifying the received credentials
[1651025910961] [28698:9495563] CHIP: [DIS] Added new fabric at index: 0x5, Initialized: 1
[1651025910961] [28698:9495563] CHIP: [DIS] Assigned compressed fabric ID: 0xF65238FEFEC78FD5, node ID: 0x000000000001B669
[1651025910961] [28698:9495563] CHIP: [CTL] Joined the fabric at index 5. Compressed fabric ID is: 0xF65238FEFEC78FD5
[1651025910961] [28698:9495563] CHIP: [CTL] Setting attestation nonce to random value
[1651025910961] [28698:9495563] CHIP: [CTL] Setting CSR nonce to random value
[1651025910961] [28698:9495563] CHIP: [CTL] Generating NOC
[1651025910962] [28698:9495563] CHIP: [DIS] Verifying the received credentials
[1651025910962] [28698:9495563] CHIP: [DIS] Added new fabric at index: 0x6, Initialized: 1
[1651025910962] [28698:9495563] CHIP: [DIS] Assigned compressed fabric ID: 0xC83705B5FE7CFCD0, node ID: 0x000000000001B669
[1651025910962] [28698:9495563] CHIP: [CTL] Joined the fabric at index 6. Compressed fabric ID is: 0xC83705B5FE7CFCD0
[1651025910994] [28698:9495567] CHIP: [DL] _OnPlatformEvent
[1651025910994] [28698:9495567] CHIP: [TOO] Sending command to group 0x4141
[1651025910994] [28698:9495567] CHIP: [TOO] Sending cluster (0x00000006) command (0x00000001) on Group 16705
[1651025910995] [28698:9495567] CHIP: [DMG] ICR moving to [AddingComm]
[1651025910995] [28698:9495567] CHIP: [DMG] ICR moving to [AddedComma]
[1651025910996] [28698:9495567] CHIP: [IN] Prepared outgoing group message 0x70000a1ac260 to 0x0000000000000000 (4)  of type 0x8 and protocolId (0, 1) on exchange 35101i with MessageCounter:100000.
[1651025910996] [28698:9495567] CHIP: [IN] Sending encrypted group msg 0x70000a1ac260 with MessageCounter:100000 to 16705 at monotonic time: 00000000263F50B0 msec to Multicast IPV6 address : UDP:[ff35:40:fd00::400:4141]:5540 with GroupID of 16705 and fabr
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the commissioner
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the controller
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the commissioner
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the controller
[1651025910996] [28698:9495563] CHIP: [IN] Expiring all connections for fabric 4!!
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the commissioner
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the controller
[1651025910996] [28698:9495563] CHIP: [IN] Expiring all connections for fabric 5!!
[1651025910996] [28698:9495563] CHIP: [CTL] Shutting down the commissioner
[... 61 lines cut ...]
[1651027035.562454][266652:266652] CHIP:CTL: Setting CSR nonce to random value
[1651027035.562609][266652:266652] CHIP:CTL: Generating NOC
[1651027035.562809][266652:266652] CHIP:DIS: Verifying the received credentials
[1651027035.563140][266652:266652] CHIP:DIS: Added new fabric at index: 0x2, Initialized: 1
[1651027035.563144][266652:266652] CHIP:DIS: Assigned compressed fabric ID: 0xB87190AE8115559F, node ID: 0x000000000001B669
[1651027035.563146][266652:266652] CHIP:CTL: Joined the fabric at index 2. Compressed fabric ID is: 0xB87190AE8115559F
[1651027035.563172][266652:266652] CHIP:CTL: Setting attestation nonce to random value
[1651027035.563176][266652:266652] CHIP:CTL: Setting CSR nonce to random value
[1651027035.563317][266652:266652] CHIP:CTL: Generating NOC
[1651027035.563496][266652:266652] CHIP:DIS: Verifying the received credentials
[1651027035.563801][266652:266652] CHIP:DIS: Added new fabric at index: 0x3, Initialized: 1
[1651027035.563805][266652:266652] CHIP:DIS: Assigned compressed fabric ID: 0xC3C4307C2268E0DE, node ID: 0x000000000001B669
[1651027035.563807][266652:266652] CHIP:CTL: Joined the fabric at index 3. Compressed fabric ID is: 0xC3C4307C2268E0DE
[1651027035.570376][266652:266657] CHIP:DL: CHIP task running
[1651027035.570411][266652:266657] CHIP:DL: HandlePlatformSpecificBLEEvent 32787
[1651027035.570418][266652:266657] CHIP:TOO: Sending command to group 0x4141
[1651027035.570420][266652:266657] CHIP:TOO: Sending cluster (0x00000006) command (0x00000001) on Group 16705
[1651027035.570430][266652:266657] CHIP:DMG: ICR moving to [AddingComm]
[1651027035.570433][266652:266657] CHIP:DMG: ICR moving to [AddedComma]
[1651027035.570565][266652:266657] CHIP:IN: Prepared outgoing group message 0x7f78637fd7f0 to 0x0000000000000000 (1)  of type 0x8 and protocolId (0, 1) on exchange 9529i with MessageCounter:90000.
[1651027035.570575][266652:266657] CHIP:IN: Sending encrypted group msg 0x7f78637fd7f0 with MessageCounter:90000 to 16705 at monotonic time: 000000013AB12B7E msec to Multicast IPV6 address : UDP:[ff35:40:fd00::100:4141]:5540 with GroupID of 16705 and fabric Id of 1
[1651027035.570595][266652:266657] CHIP:DMG: ICR moving to [CommandSen]
[1651027035.570598][266652:266657] CHIP:DMG: ICR moving to [AwaitingDe]
Segmentation fault
```

cjandhyala, 2022-04-27:
Added ACL  to the raspi  all-cluster-app using
sudo ./out/debug/chip-tool accesscontrol write acl '[{"privilege": 5, "authMode": 2, "subjects": [112233], "targets": null} , {"privilege": 4, "authMode": 3, "subjects": null, "targets": null}]' 25 0

Here is the log snippet on the raspi (all-cluster-app) after adding the above ACL and sending the _sudo ./out/debug/chip-tool onoff on 0xffffffffffff4141 1_

```

[1651101872.548033][9665:9665] CHIP:EM: Received message of type 0x8 with protocolId (0, 1) and MessageCounter:16000 on exchange 60535r
[1651101872.548116][9665:9665] CHIP:EM: Received Groupcast Message with GroupId of 16705
[1651101872.548187][9665:9665] CHIP:EM: Handling via exchange: 60535r, Delegate: 0xaaaad1a420a0
[1651101872.548282][9665:9665] CHIP:DMG: InvokeRequestMessage =
[1651101872.548333][9665:9665] CHIP:DMG: {
[1651101872.548377][9665:9665] CHIP:DMG: 	suppressResponse = false, 
[1651101872.548428][9665:9665] CHIP:DMG: 	timedRequest = false, 
[1651101872.548475][9665:9665] CHIP:DMG: 	InvokeRequests =
[1651101872.548533][9665:9665] CHIP:DMG: 	[
[1651101872.548580][9665:9665] CHIP:DMG: 		CommandDataIB =
[1651101872.548642][9665:9665] CHIP:DMG: 		{
[1651101872.548697][9665:9665] CHIP:DMG: 			CommandPathIB =
[1651101872.548759][9665:9665] CHIP:DMG: 			{
[1651101872.548818][9665:9665] CHIP:DMG: 				ClusterId = 0x6,
[1651101872.548884][9665:9665] CHIP:DMG: 				CommandId = 0x1,
[1651101872.548945][9665:9665] CHIP:DMG: 			},
[1651101872.549016][9665:9665] CHIP:DMG: 			
[1651101872.549071][9665:9665] CHIP:DMG: 			CommandData = 
[1651101872.549131][9665:9665] CHIP:DMG: 			{
[1651101872.549185][9665:9665] CHIP:DMG: 			},
[1651101872.549243][9665:9665] CHIP:DMG: 		},
[1651101872.549305][9665:9665] CHIP:DMG: 		
[1651101872.549351][9665:9665] CHIP:DMG: 	],
[1651101872.549408][9665:9665] CHIP:DMG: 	
[1651101872.549453][9665:9665] CHIP:DMG: 	InteractionModelRevision = 1
[1651101872.549498][9665:9665] CHIP:DMG: },
[1651101872.549588][9665:9665] CHIP:DMG: Received group command for Group=16705 Cluster=0x0000_0006 Command=0x0000_0001
[1651101872.550701][9665:9665] CHIP:DMG: Processing group command for Endpoint=1 Cluster=0x0000_0006 Command=0x0000_0001
[1651101872.550768][9665:9665] CHIP:DMG: AccessControl: checking f=1 a=g s=0xFFFFFFFFFFFF4141 t= c=0x0000_0006 e=1 p=o
[1651101872.550827][9665:9665] CHIP:DMG: AccessControl: allowed
[1651101872.550878][9665:9665] CHIP:ZCL: On/Off set value: 1 1
[1651101872.550927][9665:9665] CHIP:ZCL: On/off already set to new value
[1651101872.551248][9665:9665] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1651101872.551301][9665:9665] CHIP:DMG: ICR moving to [AwaitingDe]
[1651101889.032624][9665:9665] CHIP:DIS: Packet data reporter failed to parse A record
[1651101889.032807][9665:9665] CHIP:DIS: Packet data reporter failed to parse A record
[1651101889.032957][9665:9665] CHIP:DIS: Packet data reporter failed to parse A record

```

jepenven-silabs, 2022-04-27:
Was able to reproduce on Darwin setup. Investigating