project-chip/connectedhomeip #17822: Fabric Management window throws "insufficient available resources " error, once upon launch.

phonnakasturi-apple, 2022-04-27:
With SHA 99988e5ae1a2271e4eec9e0c296a3fd58358238d

Fabric Management window throws "insufficient available resources "error upon launch. This was noticed only the first time "Fabric Management" was launched.

**Steps:**

1. Install two CHIPTool apps on iOS, lets say CHIPToolApp-1 and CHIPToolApp-2
2. Pair both the CHIPTool apps
3. Open Fabric Management window on CHIPToolApp-1. 

The following error gets displayed:

**readAttributeCommissionedFabrics command failed: Error Domain=MatterInteractionErrorDomain Code=137 "An action or operation failed due to insufficient available resources." UserInfo={NSLocalizedDescription=An action or operation failed due to insufficient available resources. }.**

Note: Labels were not updated

Attaching the complete logs from M5 device.
[M5 device logs.txt](https://github.com/project-chip/connectedhomeip/files/8577511/M5.device.logs.txt)

bzbarsky-apple, 2022-04-28:
@sagar-apple how many reads does the "fabric management" screen end up issuing?

Because if we have two fabrics, both trying to do multiple reads to the device at once, and we have a cap of 4 read handlers, this will totally happen.

bzbarsky-apple, 2022-05-22:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/18505

closed 2022-05-22 by bzbarsky-apple